shivasiddharth/GassistPi #411: No clear way to uninstall

pruperting, 2018-06-22:
### Describe the bug:
```
Other than manually reading the various install scripts and manually reversing what they did, there appears to be no clear way to uninstall GassistPi. Since the install files change system files, install programs and creates autostart entries, there is quite a lot to undo.
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes
### 2. Is the issue related to assistant installation ?
No

## If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
NA
### 4. Is the assistant service restarting automatically?
NA
## If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```
NA
```

### Paste the contents of the terminal below
```

```

### Attach the log file. Log file can be found in /tmp directory. Remeber to copy your log file before shutting down or restarting the Pi.

shivasiddharth, 2018-06-24:
Please use gitter for non-issue posts.

closed 2018-06-24 by shivasiddharth

dsm320, 2019-01-10:
An uninstall tutorial would be nice. I'd rather not reflash my SD card and have to start everything from scratch

aandroide, 2019-01-10:
gassistpi is not a unique program that you can uninstall.
the only thing you can do is open the requirement folder in gassistpi open the text file and uninstall one by one the programs that are written inside the file.txt making sure that those programs do not serve to another.
programs that you no longer want to delete them with a sudo "apt-get purge program name"
finally eliminates all sdk assistant with:
pip uninstall google-assistant-sdk