shivasiddharth/GassistPi #416: Pulse audio error while playing radio

amyth1884, 2018-06-25:
### Describe the bug:
```
I searched high and low but couldnt find a soln, google assistant is accepting commands and responding as well but is not able to stream radio.
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No

## If answer to question 2 is Yes, then paste the contents of the terminal below.

```
Jun 25 23:36:39 BarryPai python[1630]: ON_CONVERSATION_TURN_STARTED
Jun 25 23:36:42 BarryPai python[1630]: ON_END_OF_UTTERANCE
Jun 25 23:36:42 BarryPai python[1630]: ON_RECOGNIZING_SPEECH_FINISHED:
Jun 25 23:36:42 BarryPai python[1630]: {"text": "play radio 2"}
Jun 25 23:36:42 BarryPai python[1630]: http://www.radiofeeds.co.uk/bbc6music.pls
Jun 25 23:36:42 BarryPai python[1630]: Tuning into Radio 2
Jun 25 23:36:46 BarryPai python[1630]: High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
Jun 25 23:36:46 BarryPai python[1630]: version 1.20.1; written and copyright by Michael Hipp and others
Jun 25 23:36:46 BarryPai python[1630]: free software (LGPL) without any warranty but with best wishes
Jun 25 23:36:46 BarryPai python[1630]: Directory: /tmp/
Jun 25 23:36:46 BarryPai python[1630]: Playing MPEG stream 1 of 1: say.mp3 ...
Jun 25 23:36:46 BarryPai python[1630]: MPEG 2.0 layer III, 32 kbit/s, 24000 Hz mono
Jun 25 23:36:47 BarryPai python[1630]: [0:01] Decoding of say.mp3 finished.
Jun 25 23:36:48 BarryPai python[1630]: [020961d8] pulse audio output error: PulseAudio server connection failure: Connection refused
Jun 25 23:36:48 BarryPai python[1630]: ON_CONVERSATION_TURN_FINISHED:
Jun 25 23:36:48 BarryPai python[1630]: {"with_follow_on_turn": false}

```

## If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
No

## If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

### Attach the log file. Log file can be found in /tmp directory. Remeber to copy your log file before shutting down or restarting the Pi.

shivasiddharth, 2018-06-25:
Please attach the log file.

amyth1884, 2018-06-29:
Logs from /tmp directory: https://pastebin.com/raw/cgdJATbV
Daemon logs : https://pastebin.com/raw/tZBfV8y1

So I tried "Play radio 5" and it started playing but I still get pulse audio errors in the logs.

Also, it would be of great help if I could trigger any shell command, I am using irrp.py to control IR devices from PI but the GPIO module in Gassistpi is interfering with it. Could you please help me remove the GPIO control from Gassistpi and instead create a function to fire shell commands such as " sudo /home/pi/irrp.py -p -g17 -f/home/pi/TV mute"

The above command works if I disable startup, I tried commenting lines to in action.py and main.py 
Any help would be highly appreciated. Thanks in advance.

shivasiddharth, 2018-06-30:
The pulse audio errors are just warnings. Please refer to one of the previous issues where i had already mentioned that. For general queries use gitter.

closed 2018-06-30 by shivasiddharth