shivasiddharth/GassistPi #417: Spotify responds with "Getting music links" and does nothing

clarkycal, 2018-06-26:
### Describe the bug:
```
Spotify responds with "Getting music links" and does nothing
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No

## If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
No

## If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
Here is the chat with me and the assistant:
```
ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play car playlist from Spotify"}
stopping vlc
Getting music links
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.23.8; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:01] Decoding of say.mp3 finished.
car playlist
1
car playlist
XXXTENTACION Jocelyn Flores

```

### Attach the log file. Log file can be found in /tmp directory. Remeber to copy your log file before shutting down or restarting the Pi.

clarkycal, 2018-06-26:
Uh, nevermind seems to be working, only with one playlist though.. Does it only work for playlists rather than just a song.

romeok01, 2018-06-27:
I have the same problem, dont play radio and spotify.

In the player.json i have:
[2, 46, "on", "Radio"]

In the .trackqueue.json  i have:
"http://stream3.polskieradio.pl:8900/listen.pls"

I write my client id and client secret i username in config.yaml file.

Google Assistant say: Thats music links and music not playing.

clarkycal, 2018-06-27:
Add a Google api-key to the config that should fix sporting as it uses YouTube

aandroide, 2018-06-27:
even with the api youtube key inserted I get the same errors as above

aandroide, 2018-06-29:
@clarkycal but in the username, did you put the name of the spotify playlist folder?

romeok01, 2018-06-30:
I have the same problem, i have playlist, user name, client_id,  client_secret and when i say: play my playlist from spotify, assistant say: thats its music links and music not play.

clarkycal, 2018-06-30:
@romeok01 It only works with YOUR playlists, not others you follow. So make a new test playlist, and call it (For example) "Test Playlist" then ask the assistant to "Play 'Test Playlist' from Spotify" should work.

romeok01, 2018-06-30:
Dont work, maybe must premium account in Spotify.

valascus, 2018-07-07:
I have a premium Spotify account, Youtube Stream works(google api works), it makes requests to Spotify, but does nothing and restart's GAssistPI.

Arul-, 2018-07-09:
I was trying my email id as the spottily username, and it did not work. It is not that clear in the instructions as well

Open Spotify music on the browser and go to your library. Here is my URL for example

    https://open.spotify.com/user/21l6p3aqb37r4t5lceuppymcq/playlist/0TQz1jpv02FCKjldReto0n

My user id is **21l6p3aqb37r4t5lceuppymcq**

Find yours and use that as your username in the `config.yaml`  and then Spotify starts working!  :)

valascus, 2018-07-09:
Thanks for your answer, Arul, but my link shows as: 
https://open.spotify.com/user/valascus/playlist/5nOnJOKadol7DcwaSLmFcb?si=Vqkl8GMkRpud7Cyz6tzQWA

So, my user is valascus?
That's what i have at config.yaml, but still no sucess

Arul-, 2018-07-09:
@valascus then that part is set right, a problem still may be elsewhere, we have to keep digging

valascus, 2018-07-09:
It makes requests... I can see in the graphs from Spotify developers page, but... Same issue.

Arul-, 2018-07-10:
@valascus upon further digging, I found the following help

- [ ] make sure your Spotify playlist is public and 
- [ ] it's not muted in the web player

valascus, 2018-07-10:
Thank you once again.
Both of that requeriments were and are checked.

valascus, 2018-07-11:
Where did you get that "username"
My "username", as far as i know, is valascus. That's what appears at profile and when i generate a link for a playlist.

aandroide, 2018-07-12:
the link is: https://www.spotify.com/it/account/overview/
to get the username once logged in you must then copy it to config.yaml

aandroide, 2018-07-12:
@valascus verify the user name better, usually they are numbers and letters or numbers only or try to register in spotify with the same email you use for gassistpi and create a new app as explained in the readme and then access the link I mentioned above. As soon as I logged in, the page with the username in numbers appeared. can you attach a screenshot?

valascus, 2018-07-12:
None of them worked.
In fact, GAssistPi didn't restart, which is good, but not playing anything tho.
Here's a screenshot of my spotify profile: https://prnt.sc/k5qdvv

valascus, 2018-07-12:
It is valascus. It get's the Playlist. It says "1" (position where the playlist is when getting it). It was 5 and 6 before. I just deleted all the other playlists. It get's data. With the "users" u said, it gives 0.

aandroide, 2018-07-12:
At this point I would try to redo the registration to spotify from scratch

valascus, 2018-07-12:
But i need premium to use this feature, which i would not have if i changed my account.

aandroide, 2018-07-12:
@valascus I have no premium and it works great.

valascus, 2018-07-12:
Rly? 
I thought u needed to be premium for spotify developers dashboard. 
I'll try it in a sec.

aandroide, 2018-07-12:
let's go by exclusion:
but does the strem from youtube work for you? have you enabled the bees from the console?

valascus, 2018-07-12:
Doesnt work with a new account either. As in the main account, it does find the playlist, but nothing else.

Youtube Stream works fine.
Enable the bees?? What's that??

valascus, 2018-07-12:
If u mean API's, yeah. i did. As i just said, youtube works.

aandroide, 2018-07-12:
ok 
when you edit the config.yaml file, stop the assistant and restart to activate the new changes did you do it?

valascus, 2018-07-12:
Of course.

aandroide, 2018-07-12:
ok, can you attach your actions.py file and your config.yaml?

valascus, 2018-07-12:
[Spotify.zip](https://github.com/shivasiddharth/GassistPi/files/2189903/Spotify.zip)

I let my personal configs there so u can actually test it.

aandroide, 2018-07-12:
I have pasted your files in my project and are playing:
ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "Play best playlist from Spotify"}
stopping vlc
Getting music links
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.23.8; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:01] Decoding of say.mp3 finished.
best playlist
1
best playlist
The Chainsmokers Something Just Like This
[02ac6970] pulse audio output error: PulseAudio server connection failure: Connection refused
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}

aandroide, 2018-07-12:
ok google Play best playlist from Spotify

valascus, 2018-07-12:
So, the problem is not at the configuration itself. What could it be?

valascus, 2018-07-12:
As you can see, nothing special here...

`Jul 12 18:39:09 raspberrypi python[12166]: ON_END_OF_UTTERANCE
Jul 12 18:39:10 raspberrypi python[12166]: ON_RECOGNIZING_SPEECH_FINISHED:
Jul 12 18:39:10 raspberrypi python[12166]:   {"text": "play best playlist on Spotify"}
Jul 12 18:39:10 raspberrypi python[12166]: stopping vlc
Jul 12 18:39:10 raspberrypi python[12166]: Getting music links
Jul 12 18:39:11 raspberrypi python[12166]: High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
Jul 12 18:39:11 raspberrypi python[12166]: #011version 1.23.8; written and copyright by Michael Hipp and others
Jul 12 18:39:11 raspberrypi python[12166]: #011free software (LGPL) without any warranty but with best wishes
Jul 12 18:39:12 raspberrypi python[12166]: Directory: /tmp/
Jul 12 18:39:12 raspberrypi python[12166]: Playing MPEG stream 1 of 1: say.mp3 ...
Jul 12 18:39:12 raspberrypi python[12166]: MPEG 2.0 L III cbr32 24000 mono
Jul 12 18:39:13 raspberrypi kernel: [785813.980735] Voltage normalised (0x00000000)
Jul 12 18:39:13 raspberrypi python[12166]: [0:01] Decoding of say.mp3 finished.
Jul 12 18:39:13 raspberrypi python[12166]: best playlist on spotify
Jul 12 18:39:14 raspberrypi python[12166]: 1
Jul 12 18:39:14 raspberrypi python[12166]: best playlist
Jul 12 18:39:14 raspberrypi python[12166]: ON_CONVERSATION_TURN_FINISHED:
Jul 12 18:39:14 raspberrypi python[12166]:   {"with_follow_on_turn": false}`

aandroide, 2018-07-12:
not on but from spotify

valascus, 2018-07-12:
Shouldn't mather, as he recognizes it.
But w8

valascus, 2018-07-12:
No. Same. Issue.

aandroide, 2018-07-12:
the configuration is correct.
you can attach
gassistpi.log that find folder temp

aandroide, 2018-07-12:
I do not like this message:
 raspberrypi kernel: [785813.980735] Voltage normalised (0x00000000) 

have you tried to change power supply?
it could go haywire with voltage drops

valascus, 2018-07-12:
For some reason, it worked now.
Yes, i need to change it. The one that i bought should be fitting the Raspberry PI just fine, as i bought it just for that purpose. It was announced as "for RaspBerry Pi 3". It has an imbebbed Switch to turn on/off the PI, but only 5V, and 2.5A

aandroide, 2018-07-12:
if the problem should come back replace the power supply. the scripts are ok

sanderthierens, 2018-07-13:
hi
i have a problem to. I have a chrome cast called bedroom tv and when i say " hey google, play spotify on my bedroom tv" it says " getting music links" but my assistant on my raspberry pi restarts itsef and I cant ask her things for a short while until I hear the start sound off the assistant. spotify plays normal. alo that " getting music links" is rally annoying to can I turn it off?

valascus, 2018-07-13:
You can't say that sentence. It won't recognize.
You must say: Ok Google, Play "NAME OF THE PLAYLIST" from Spotify.
Only that.

It only works with YOUR playlists.

sanderthierens, 2018-07-13:
oke, but I still have that problem that my assistant stops working and restart it self after I say Ok Google, Play "NAME OF THE PLAYLIST" from Spotify.  And it is only with spotify not when i say Ok Google, play " name of a song" on youtube. And can I turn off the words she says "getting music links"   every time i ask her to play something. also I cant find the config.yaml file.

ps: I am new to this world, so my apologies when I ask silly questions.

valascus, 2018-07-13:
ok. That's your problem then. You can't find config.yaml?
If you followed the intructions, it should be at /home/pi/GassistPi/src/config.yaml.

You must setup a google api to get from youtube and define your spotify credentials.

joerivd1, 2018-07-16:
hallo I have the same problem as sanderthierens. afther I say Ok Google, Play "NAME OF THE PLAYLIST" from Spotify. the assistant responds with getting music links and plays the playlist but afther it says getting music liks? my assistant don't respond to ok google or hey google for a small time and then i hear the same sound when I turn on the raspberry pi 3 ( the sound that the assistant is starting and ends with "hi"). so i assume that the assistant crashes and itself restart. afther that sound i can start to ask things again. can somone help me with this problem?

shivasiddharth, 2018-07-16:
@sanderthierens and @joerivd1 It can mean only one thing, the spotify credentials have not been entered properly in the config.yaml.

joerivd1, 2018-07-16:
@shivasiddharth  thanks it worked for me I made a small mistake with the cllient secret. one more question? is it posible to not hear the ' getting music links' ?

shivasiddharth, 2018-07-16:
i have intentionally placed it to know when error is happening. if you dont want to hear that, search for a line that starts with "say" within the functions assigned for spotify and comment that.

joerivd1, 2018-07-16:
how do you mean exactly with functions assigned for spotify? where can i find that

aandroide, 2018-07-16:
@joerivd1 in actions.py comment on the line 1303

joerivd1, 2018-07-16:
ef spotify_playlist_select(phrase):
    trackslist=[]
    currenttrackid=0
    idx=phrase.find('play')
    track=phrase[idx:]
    track=track.replace("'}", "",1)
    track = track.replace('play','',1)
    track = track.replace('from spotify','',1)
    track=track.strip()
    say("Getting music links")
    print(track)
    playlists,num=scan_spotify_playlists()
    if not num==[]:
        for i in range(0,num):
            print(str(playlists['Playlists'][i][1]).lower())
            if track in str(playlists['Playlists'][i][1]).lower():
                trackslist=playlists['Playlists'][i][2]
                break
        if not trackslist==[]:
            vlcplayer.media_manager(trackslist,'Spotify')
            vlcplayer.spotify_player(currenttrackid)
    else:
        say("Unable to find matching playlist")

shivasiddharth, 2018-07-17:
@joerivd1 , comment those lines with `say` if you do not want to hear voice prompts. You can comment it by adding a # symbol at the start of the line.

sanderthierens, 2018-07-17:
@shivasiddharth  so how do I do it properly in the config.yaml.
I have now and the username is correct from spotify

#Spotify credentials. Register for a spotify developer and get the credentials.
Spotify:
  client_id: 7405d02b211a4163ae21fcfbdbba0482
  client_secret: 5e6bb44c4e084fbe9e3f42eb7901819d
  username: 11169079112

I checked again and everyting is correct. and problem is still happening
that the assistant restart it self

shivasiddharth, 2018-07-17:
@sanderthierens , Two things. 
1.The credentials should be entered within the quotes just like how you had them in the unedited config.yaml file. 
2. I am not sure, if spotify assigns usernames with numbers (but i could be wrong), so double check your spotify username in https://www.spotify.com/account/overview/.

sanderthierens, 2018-07-17:
@shivasiddharth I was not sure eather before but I asked and is was indeed my username, sotify generate the username themselve now, no idea why. do I need to restart the pi3 afther I changed somthing in config.yaml file?

aandroide, 2018-07-17:
@shivasiddharth 
I confirm that with a Free account, spotify assigns a username that contains numbers and letters

shivasiddharth, 2018-07-17:
@sanderthierens , you need to restart the assistant. Restart of Pi is not required.

oscarrc, 2018-07-20:
Same issue here, I've checked my username, api keys and the playlist is not muted on the web player (as someone suggested before). The music does not play (Youtube streaming works fine).

```
ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play party playlist from Spotify"}
stopping vlc
Obtener enlaces de música
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
        version 1.23.8; written and copyright by Michael Hipp and others
        free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:02] Decoding of say.mp3 finished.
party playlist
5
party
favoritos
favoritos
favoritos
favoritos
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}
```

[3 more replies]

closed 2018-08-10 by shivasiddharth