shivasiddharth/GassistPi #421: execute a program

kumarsatish23, 2018-06-29:
### Describe the bug:
```

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No

## If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
Yes

## If answer to question 4 is Yes, then paste the contents of the terminal below.

```
i don't say it as a bug the thing is that i want to execute a program (say"abc.py") from a particular path when i speak out a key word
```

## If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

### Attach the log file. Log file can be found in /tmp directory. Remeber to copy your log file before shutting down or restarting the Pi.

kumarsatish23, 2018-06-29:
help me

aandroide, 2018-06-29:
use the chat for these non-installation questions 
https://gitter.im/GassistPi/Lobby/

shivasiddharth, 2018-06-30:
Please use gitter for general queries. Closing this.

closed 2018-06-30 by shivasiddharth