shivasiddharth/GassistPi #427: Attempting to Play

TheEndernaut, 2018-07-04:
### Describe the bug:
```When I I try to play a spotify playlist, The Assistant will stop before music plays

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No

## If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
No

## If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```
Getting music links
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.23.8; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:01] Decoding of say.mp3 finished.
cast music

```

### Attach the log file. Log file can be found in /tmp directory. Remeber to copy your log file before shutting down or restarting the Pi.

[GassistPi.log](https://github.com/shivasiddharth/GassistPi/files/2162155/GassistPi.log)

aandroide, 2018-07-05:
#417

shivasiddharth, 2018-07-06:
Cn you please describe the bug, just the logs, gives very little info and it takes a lot of time for me to figure out the issue from logs without bug description.

TheEndernaut, 2018-07-06:
The issue is just that when I request a Spotify playlist, the GassistPi task ends. It gives no output, and it generally comes by surprise. Thank you so much for all your work on this project!

aandroide, 2018-07-11:
I solved the problem too. It was in the username that I thought was the one created in the app of the phone and instead accessing to spotify from the pc and accessing with the credentials I discovered that the username was totally different, I then copied the username in config.yaml and everything works now

shivasiddharth, 2018-07-17:
@TheEndernaut , Looks like a spotify credentials declaration issue.  Please go through #417. If that does not fox your issue you can get back here.

shivasiddharth, 2018-08-10:
Closing this for inactivity

closed 2018-08-10 by shivasiddharth