shivasiddharth/GassistPi #432: Cannot complete install sudo  /home/pi/GassistPi/scripts/gassist-installer-pi-zero.sh

doomdoomydoomed, 2018-07-07:
### Describe the bug:

When finishing the install I get the following error. It is showing a 404 error but when checking the repo the file is located and reachable. Can this file be manually installed and rerun the installer to complete it?

Collecting python-vlc>=3.0.102 (from -r /home/pi/GassistPi/Requirements/GassistPi-pip-requirements.txt (line 18))
  Using cached https://www.piwheels.org/simple/python-vlc/python_vlc-3.0.102-py3-none-any.whl
Collecting pyyaml>=3.12 (from -r /home/pi/GassistPi/Requirements/GassistPi-pip-requirements.txt (line 20))
  HTTP error 404 while getting https://www.piwheels.org/simple/pyyaml/PyYAML-4.1-cp35-cp35m-linux_armv7l.whl#sha256=a5b125a24244830d574ccfd8fab861198d2ffe491b73179ffd75abf78caee6ff (from https://www.piwheels.org/simple/pyyaml/)
  Could not install requirement pyyaml>=3.12 from https://www.piwheels.org/simple/pyyaml/PyYAML-4.1-cp35-cp35m-linux_armv7l.whl#sha256=a5b125a24244830d574ccfd8fab861198d2ffe491b73179ffd75abf78caee6ff (from -r /home/pi/GassistPi/Requirements/GassistPi-pip-requirements.txt (line 20)) because of error 404 Client Error: Not Found for url: https://www.piwheels.org/simple/pyyaml/PyYAML-4.1-cp35-cp35m-linux_armv7l.whl
Could not install requirement pyyaml>=3.12 from https://www.piwheels.org/simple/pyyaml/PyYAML-4.1-cp35-cp35m-linux_armv7l.whl#sha256=a5b125a24244830d574ccfd8fab861198d2ffe491b73179ffd75abf78caee6ff (from -r /home/pi/GassistPi/Requirements/GassistPi-pip-requirements.txt (line 20)) because of HTTP error 404 Client Error: Not Found for url: https://www.piwheels.org/simple/pyyaml/PyYAML-4.1-cp35-cp35m-linux_armv7l.whl for URL https://www.piwheels.org/simple/pyyaml/PyYAML-4.1-cp35-cp35m-linux_armv7l.whl#sha256=a5b125a24244830d574ccfd8fab861198d2ffe491b73179ffd75abf78caee6ff (from https://www.piwheels.org/simple/pyyaml/)

```

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes/No

### 2. Is the issue related to assistant installation ?
Yes/No

## If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes/No

### 4. Is the assistant service restarting automatically?
Yes/No

## If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

### Attach the log file. Log file can be found in /tmp directory. Remeber to copy your log file before shutting down or restarting the Pi.

Arul-, 2018-07-08:
I'm also getting similar error and the installation stops

While running `sudo  /home/pi/GassistPi/scripts/gassist-installer-pi3.sh` I get the following

```
HTTP error 404 while getting https://www.piwheels.org/simple/pyyaml/PyYAML-4.1-cp35-cp35m-linux_armv7l.whl#sha256=a5b125a24244830d574ccfd8fab861198d2ffe491b73179ffd75abf78caee6ff (from https://www.piwheels.org/simple/pyyaml/)
  Could not install requirement pyyaml>=3.12 from https://www.piwheels.org/simple/pyyaml/PyYAML-4.1-cp35-cp35m-linux_armv7l.whl#sha256=a5b125a24244830d574ccfd8fab861198d2ffe491b73179ffd75abf78caee6ff (from -r /home/pi/GassistPi/Requirements/GassistPi-pip-requirements.txt (line 20)) because of error 404 Client Error: Not Found for url: https://www.piwheels.org/simple/pyyaml/PyYAML-4.1-cp35-cp35m-linux_armv7l.whl
Could not install requirement pyyaml>=3.12 from https://www.piwheels.org/simple/pyyaml/PyYAML-4.1-cp35-cp35m-linux_armv7l.whl#sha256=a5b125a24244830d574ccfd8fab861198d2ffe491b73179ffd75abf78caee6ff (from -r /home/pi/GassistPi/Requirements/GassistPi-pip-requirements.txt (line 20)) because of HTTP error 404 Client Error: Not Found for url: https://www.piwheels.org/simple/pyyaml/PyYAML-4.1-cp35-cp35m-linux_armv7l.whl for URL https://www.piwheels.org/simple/pyyaml/PyYAML-4.1-cp35-cp35m-linux_armv7l.whl#sha256=a5b125a24244830d574ccfd8fab861198d2ffe491b73179ffd75abf78caee6ff (from https://www.piwheels.org/simple/pyyaml/)
```

Bojc, 2018-07-08:
HI.
I am also getting this error on raspberry pi 3. Few days ago it worked just fine. Now i can't go past it. It is always the same error:
HTTP error 404 while getting https://www.piwheels.org/simple/pyyaml/PyYAML-4.1-cp35-cp35m-linux_armv7l.whl#sha256=a5b125a24244830d574ccfd8fab861198d2ffe491b73179ffd75abf78caee6ff (from https://www.piwheels.org/simple/pyyaml/)
  Could not install requirement pyyaml>=3.12 from https://www.piwheels.org/simple/pyyaml/PyYAML-4.1-cp35-cp35m-linux_armv7l.whl#sha256=a5b125a24244830d574ccfd8fab861198d2ffe491b73179ffd75abf78caee6ff (from -r /home/pi/GassistPi/Requirements/GassistPi-pip-requirements.txt (line 20)) because of error 404 Client Error: Not Found for url: https://www.piwheels.org/simple/pyyaml/PyYAML-4.1-cp35-cp35m-linux_armv7l.whl
Could not install requirement pyyaml>=3.12 from https://www.piwheels.org/simple/pyyaml/PyYAML-4.1-cp35-cp35m-linux_armv7l.whl#sha256=a5b125a24244830d574ccfd8fab861198d2ffe491b73179ffd75abf78caee6ff (from -r /home/pi/GassistPi/Requirements/GassistPi-pip-requirements.txt (line 20)) because of HTTP error 404 Client Error: Not Found for url: https://www.piwheels.org/simple/pyyaml/PyYAML-4.1-cp35-cp35m-linux_armv7l.whl for URL https://www.piwheels.org/simple/pyyaml/PyYAML-4.1-cp35-cp35m-linux_armv7l.whl#sha256=a5b125a24244830d574ccfd8fab861198d2ffe491b73179ffd75abf78caee6ff (from https://www.piwheels.org/simple/pyyaml/)
 
It is in step : 4.2 "Execute the installers" in the guide for google assistant.

pierremtb, 2018-07-08:
For some reason the last stable version of PyYAML, 4.1, is no more available on the piwheels website. However, since the script only needs a dependency newer or equal to 3.12, before running step 4.2, one can easily edit the last line of the file `Requirements/GassistPi-pip-requirements.txt`, replacing `pyyaml>=3.12` with `pyyaml==3.13` for instance, since the 3.13 doesn't trigger a 404 error on piwheels.

Arul-, 2018-07-08:
I found a solution by installing pyYAML globally with `sudo apt-get install python-yaml` and removing the last line in `/home/pi/GassistPi/Requirements/GassistPi-pip-requirements.txt`

But @pierremtb already posted a better solution!

Bojc, 2018-07-08:
I will try now. So only thing that i have to do is change the last line in "Requirements/GassistPi-pip-requirements.txt" file,  from  pyyaml>=3.12  to pyyaml=3.13, and run: 
1. sudo chmod +x /home/pi/GassistPi/scripts/gassist-installer-pi3.sh and then
2. sudo  /home/pi/GassistPi/scripts/gassist-installer-pi3.sh ?

Bojc, 2018-07-08:
Thank you both for help. It works now. You guys made my day. Thanks again.

shivasiddharth, 2018-07-12:
Issue has been resolved.

closed 2018-07-12 by shivasiddharth