shivasiddharth/GassistPi #433: MUSIC STREAMING FROM YOUTUBE

Bojc, 2018-07-08:
Hi. 
I have a problem with MUSIC STREAMING FROM YOUTUBE. In new actions.py script, i can't find location to put my API key in. When i say "stream I got you" assistant sayes " that is a youtube link" or something like that, and then it reboots. The same problem is with streming radio. I can't find location to put in streaming links. When i say "play radio 5" it plays, but i don't know which station.
I tryed a few things, but with no luck.
Can someone help?

Arul-, 2018-07-10:
Take a look at `config.yaml` in the same folder. Once I updated that mine started working. But still when I ask for something that it could not find, it does crash and restart

Bojc, 2018-07-10:
Hi.
I managed to get my radio to work now. But I still can't get Youtube to work. I copied my API  in config.yaml wher it says:
#API Key for YouTube and Kickstarter Search Engine
Google_cloud_api_key: AIzaSyAw.................................ylys
When I say  for example "stream stairway to heaven", it says "getting Youtube link" or something like that, and then it reboots. I tried several strem comands, but with no luck.

Best regards.

Bojc, 2018-07-10:
Ok. I solved the problem now. I looked at my API key settings again in Google cloud platform under Key restrictions. In API restrictions, there was a "BigQuery API" restriction set by default. I removed that restriction and now it works fine. Thank you again for your help.

closed 2018-07-12 by shivasiddharth