shivasiddharth/GassistPi #435: Does not play sound

ericdraven22, 2018-07-09:
Sorry for the translation, but I do not speak English very well.
I'm trying to play the radio or spotify and nobody walks me, it does not give me an error, it just does not do anything.
If I say, play radio one, Tuning into Radio 1 tells me, it recognizes the link of the radio but does not play anything.
Here below I paste what it shows per terminal and the log.

If you can help me, I thank you very much.

N_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play Radio 1"}
http://provisioning.streamtheworld.com/pls/LOS40_ARGENTINA.pls
Tuning into Radio 1
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.23.8; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:01] Decoding of say.mp3 finished.

----------------------------------------------------------------------------------------------------------------
log
2018-07-09 20:11:02,805 DEBUG urllib3.connectionpool Starting new HTTPS connection (1): translate.google.com:443
2018-07-09 20:11:03,326 DEBUG urllib3.connectionpool https://translate.google.com:443 "GET / HTTP/1.1" 200 None
2018-07-09 20:11:03,706 DEBUG urllib3.connectionpool https://translate.google.com:443 "GET /translate_a/single?hl=en&client=t&q=Tuning+into+Radio+1&ie=UTF-8&tsel=0&tl=en&sl=auto&oe=UTF-8&otf=1&tk=148692.276923&ssel=0&dt=at&dt=bd&dt=ex&dt=ld&dt=md&dt=qca&dt=rw&dt=rm&dt=ss&dt=t HTTP/1.1" 200 None
2018-07-09 20:11:04,098 DEBUG gtts.tts tokenizer_func: <bound method Tokenizer.run of re.compile('(?<=\\?).|(?<=\\!).|(?<=\\？).|(?<=\\！).|(?<!\\.[a-z])\\. |(?<!\\.[a-z])\\, |\\)|\\\n|\\—|\\：|\\;|\\(|\\、|\\，|\\[|\\¿|\\:|\\،|\\‥|\\¡|\\]|\\…|\\。', re.IGNORECASE) from: [<function tone_marks at 0x72596c00>, <function period_comma at 0x72596c48>, <function other_punctuation at 0x72596c90>]>
2018-07-09 20:11:04,099 DEBUG gtts.tts pre_processor_funcs: [<function tone_marks at 0x725966f0>, <function end_of_line at 0x72596ae0>, <function abbreviations at 0x72596b28>, <function word_sub at 0x72596b70>]
2018-07-09 20:11:04,101 DEBUG gtts.tts lang: en
2018-07-09 20:11:04,101 DEBUG gtts.tts lang_check: True
2018-07-09 20:11:04,102 DEBUG gtts.tts text: Tuning into Radio 1
2018-07-09 20:11:04,103 DEBUG gtts.tts slow: False
2018-07-09 20:11:04,119 DEBUG urllib3.connectionpool Starting new HTTP connection (1): translate.google.com:80
2018-07-09 20:11:04,332 DEBUG urllib3.connectionpool http://translate.google.com:80 "GET / HTTP/1.1" 302 226
2018-07-09 20:11:04,370 DEBUG urllib3.connectionpool Starting new HTTPS connection (1): translate.google.com:443
2018-07-09 20:11:04,731 DEBUG urllib3.connectionpool https://translate.google.com:443 "GET / HTTP/1.1" 200 None
2018-07-09 20:11:04,935 DEBUG urllib3.connectionpool Starting new HTTP connection (1): translate.google.com:80
2018-07-09 20:11:04,996 DEBUG urllib3.connectionpool http://translate.google.com:80 "GET //translate/releases/twsfe_w_20180702_RC00/r/js/desktop_module_main.js HTTP/1.1" 200 163005
2018-07-09 20:11:05,172 DEBUG gtts.tts pre-processing: <function tone_marks at 0x725966f0>
2018-07-09 20:11:05,176 DEBUG gtts.tts pre-processing: <function end_of_line at 0x72596ae0>
2018-07-09 20:11:05,177 DEBUG gtts.tts pre-processing: <function abbreviations at 0x72596b28>
2018-07-09 20:11:05,183 DEBUG gtts.tts pre-processing: <function word_sub at 0x72596b70>
2018-07-09 20:11:05,184 DEBUG gtts.tts text_parts: 1
2018-07-09 20:11:05,191 DEBUG urllib3.connectionpool Starting new HTTPS connection (1): translate.google.com:443
2018-07-09 20:11:05,529 DEBUG urllib3.connectionpool https://translate.google.com:443 "GET / HTTP/1.1" 200 None
2018-07-09 20:11:05,548 DEBUG gtts.tts payload-0: {'textlen': 19, 'client': 'tw-ob', 'idx': 0, 'tk': '148692.276923', 'total': 1, 'q': 'Tuning into Radio 1', 'ie': 'UTF-8', 'tl': 'en', 'ttsspeed': 1}
2018-07-09 20:11:05,557 DEBUG urllib3.connectionpool Starting new HTTPS connection (1): translate.google.com:443
2018-07-09 20:11:06,013 DEBUG urllib3.connectionpool https://translate.google.com:443 "GET /translate_tts?textlen=19&client=tw-ob&idx=0&tk=148692.276923&total=1&q=Tuning+into+Radio+1&ie=UTF-8&tl=en&ttsspeed=1 HTTP/1.1" 200 7968
2018-07-09 20:11:06,020 DEBUG gtts.tts headers-0: {'User-Agent': 'Mozilla/5.0 (Windows NT 10.0; WOW64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/47.0.2526.106 Safari/537.36', 'Accept': '*/*', 'Referer': 'http://translate.google.com/', 'Accept-Encoding': 'gzip, deflate', 'Connection': 'keep-alive'}
2018-07-09 20:11:06,021 DEBUG gtts.tts url-0: https://translate.google.com/translate_tts?textlen=19&client=tw-ob&idx=0&tk=148692.276923&total=1&q=Tuning+into+Radio+1&ie=UTF-8&tl=en&ttsspeed=1
2018-07-09 20:11:06,022 DEBUG gtts.tts status-0: 200
2018-07-09 20:11:06,023 DEBUG gtts.tts part-0 written to <_io.BufferedWriter name='/tmp/say.mp3'>
2018-07-09 20:11:06,024 DEBUG gtts.tts Saved to /tmp/say.mp3

shivasiddharth, 2018-07-12:
Can you please try changing the media volume to maximum using the syntaxes given in the readme.

ericdraven22, 2018-07-12:
Thanks for answering.
That's the first thing I probe. I followed the following instructions:

After successful authentication, the Google Assistant Demo test will automatically start. At the start, the volume might be low, the assistant volume is independent of the Pi volume, so increase the volume by using "Hey Google, Set volume to maximum" command.

But it did not work, after indicating the action to google does not answer anything, but i look at the code that comes out by terminal and gives me the answer "alright"

ericdraven22, 2018-07-13:
Yesterday I tried to change the playback mode from USB to HDMI and it did not work either.
He also tried to say "set volume up" and he answered that the volume was already as high as possible.
Try to play from spotify watching the output in the terminal, I say play ... on spotify and I recognize everything, it brings me the two playlist I have created, in the log I see that sends to play the first song of the playlist that you indicate but nothing is heard either.
Will I have a problem with the vlc? with the installation?

ericdraven22, 2018-07-15:
Hello!
I was able to solve the problems, I comment them in case it helps someone.
The problem with the radio was that the link had extension .pls and those do not work, try with another one that has .mp3 extension and works perfect.
The problem with spotify was that I had not put the google cloud api on YouTube, I put that in and it went on.

closed 2018-07-15 by ericdraven22

elatrium, 2018-07-28:
I had the same problem. Please change the  default yaml file with working stations!