shivasiddharth/GassistPi #440: rpi zero w setup issue

rohit366, 2018-07-12:
Hi I have followed the instruction as per your readme guild for rpi zero w but I am getting an error on the last step.

all the above steps have been finished successfully. here is the log for the error I am getting

pi@raspberrypi:~/env/bin $ python3 -u /home/pi/GassistPi/src/pushbutton.py --project-id rpi-assistant-207815 --device-model-id rpi-assistant-207815-rpi-zero-fzkpjy
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/pushbutton.py", line 30, in <module>
    import grpc
ImportError: No module named 'grpc'
pi@raspberrypi:~/env/bin $ python -u /home/pi/GassistPi/src/pushbutton.py --project-id rpi-assistant-207815 --device-model-id rpi-assistant-207815-rpi-zero-fzkpjy
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/pushbutton.py", line 35, in <module>
    from actions import say
  File "/home/pi/GassistPi/src/actions.py", line 1216
SyntaxError: Non-ASCII character '\xc3' in file /home/pi/GassistPi/src/actions.py on line 1216, but no encoding declared; see http://python.org/dev/peps/pep-0263/ for details

rohit366, 2018-07-12:
resolved the issue by adding utf-8 encoding at the top of the actions.py file.

shivasiddharth, 2018-07-16:
Please use the issue template.

closed 2018-07-16 by shivasiddharth