shivasiddharth/GassistPi #441: problem in Youtube streaming

Shidiqadm, 2018-07-16:
Hi sid thank you for this wonderful work. iam using gassist pi for recent 3 months i havent faced any noticible error but now when i say Hey google "stream sia cheap thrills" it says "sorry i cannot understand",when i used to start manually i noticed that whenever i say stream it recognizes as 'dream' or 'trim' its hardly working sometimes.
is there any way to change stream keyword
And iam attaching my log files 

2018-07-16 18:07:12,797 DEBUG google.auth.transport.requests Making request: POST https://accounts.google.com/o/oauth2/token
2018-07-16 18:07:12,810 DEBUG urllib3.connectionpool Starting new HTTPS connection (1): accounts.google.com:443
2018-07-16 18:07:14,317 DEBUG urllib3.connectionpool https://accounts.google.com:443 "POST /o/oauth2/token HTTP/1.1" 200 None
2018-07-16 18:17:37,356 DEBUG urllib3.connectionpool Starting new HTTPS connection (1): translate.google.com:443
2018-07-16 18:17:38,048 DEBUG urllib3.connectionpool https://translate.google.com:443 "GET / HTTP/1.1" 200 None
2018-07-16 18:17:38,254 DEBUG urllib3.connectionpool https://translate.google.com:443 "GET /translate_a/single?ssel=0&ie=UTF-8&tl=en&q=Getting+youtube+link&tsel=0&otf=1&hl=en&client=t&sl=auto&tk=963959.576379&dt=at&dt=bd&dt=ex&dt=ld&dt=md&dt=qca&dt=rw&dt=rm&dt=ss&dt=t&oe=UTF-8 HTTP/1.1" 200 None
2018-07-16 18:17:38,261 DEBUG gtts.tts tokenizer_func: <bound method Tokenizer.run of re.compile('(?<=\\?).|(?<=\\!).|(?<=\\？).|(?<=\\！).|(?<!\\.[a-z])\\. |(?<!\\.[a-z])\\, |\\(|\\：|\\，|\\。|\\¿|\\—|\\[|\\)|\\‥|\\\n|\\،|\\;|\\]|\\¡|\\…|\\、|\\:', re.IGNORECASE) from: [<function tone_marks at 0x726a7738>, <function period_comma at 0x726a7780>, <function other_punctuation at 0x726a77c8>]>
2018-07-16 18:17:38,262 DEBUG gtts.tts lang: en
2018-07-16 18:17:38,263 DEBUG gtts.tts lang_check: True
2018-07-16 18:17:38,263 DEBUG gtts.tts slow: False
2018-07-16 18:17:38,263 DEBUG gtts.tts text: Getting youtube link
2018-07-16 18:17:38,263 DEBUG gtts.tts pre_processor_funcs: [<function tone_marks at 0x726a7228>, <function end_of_line at 0x726a7618>, <function abbreviations at 0x726a7660>, <function word_sub at 0x726a76a8>]
2018-07-16 18:17:38,271 DEBUG urllib3.connectionpool Starting new HTTP connection (1): translate.google.com:80
2018-07-16 18:17:39,357 DEBUG urllib3.connectionpool http://translate.google.com:80 "GET / HTTP/1.1" 301 199
2018-07-16 18:17:39,572 DEBUG urllib3.connectionpool http://translate.google.com:80 "GET /m HTTP/1.1" 302 227
2018-07-16 18:17:39,593 DEBUG urllib3.connectionpool Starting new HTTPS connection (1): translate.google.com:443
2018-07-16 18:17:40,038 DEBUG urllib3.connectionpool https://translate.google.com:443 "GET /m HTTP/1.1" 200 None
2018-07-16 18:17:40,105 DEBUG gtts.tts Unable to get language list: 'NoneType' object is not subscriptable
Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/gtts/lang.py", line 36, in tts_langs
    langs.update(_fetch_langs())
  File "/home/pi/env/lib/python3.5/site-packages/gtts/lang.py", line 62, in _fetch_langs
    js_path = soup.find(src=re.compile(JS_FILE))['src']
TypeError: 'NoneType' object is not subscriptable

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/gtts/tts.py", line 117, in __init__
    langs = tts_langs()
  File "/home/pi/env/lib/python3.5/site-packages/gtts/lang.py", line 41, in tts_langs
    raise RuntimeError("Unable to get language list: %s" % str(e))
RuntimeError: Unable to get language list: 'NoneType' object is not subscriptable
2018-07-16 18:17:40,113 WARNING gtts.tts Unable to get language list: 'NoneType' object is not subscriptable
2018-07-16 18:17:40,114 DEBUG gtts.tts pre-processing: <function tone_marks at 0x726a7228>
2018-07-16 18:17:40,118 DEBUG gtts.tts pre-processing: <function end_of_line at 0x726a7618>
2018-07-16 18:17:40,119 DEBUG gtts.tts pre-processing: <function abbreviations at 0x726a7660>
2018-07-16 18:17:40,126 DEBUG gtts.tts pre-processing: <function word_sub at 0x726a76a8>
2018-07-16 18:17:40,128 DEBUG gtts.tts text_parts: 1
2018-07-16 18:17:40,136 DEBUG urllib3.connectionpool Starting new HTTPS connection (1): translate.google.com:443
2018-07-16 18:17:40,552 DEBUG urllib3.connectionpool https://translate.google.com:443 "GET / HTTP/1.1" 200 None
2018-07-16 18:17:40,577 DEBUG gtts.tts payload-0: {'textlen': 20, 'ie': 'UTF-8', 'tl': 'en', 'tk': '963959.576379', 'q': 'Getting youtube link', 'total': 1, 'ttsspeed': 1, 'idx': 0, 'client': 'tw-ob'}
2018-07-16 18:17:40,586 DEBUG urllib3.connectionpool Starting new HTTPS connection (1): translate.google.com:443
2018-07-16 18:17:41,008 DEBUG urllib3.connectionpool https://translate.google.com:443 "GET /translate_tts?textlen=20&ie=UTF-8&tl=en&tk=963959.576379&q=Getting+youtube+link&total=1&ttsspeed=1&idx=0&client=tw-ob HTTP/1.1" 200 5856
2018-07-16 18:17:41,015 DEBUG gtts.tts headers-0: {'Accept-Encoding': 'gzip, deflate', 'Accept': '*/*', 'User-Agent': 'Mozilla/5.0 (Windows NT 10.0; WOW64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/47.0.2526.106 Safari/537.36', 'Connection': 'keep-alive', 'Referer': 'http://translate.google.com/'}
2018-07-16 18:17:41,017 DEBUG gtts.tts url-0: https://translate.google.com/translate_tts?textlen=20&ie=UTF-8&tl=en&tk=963959.576379&q=Getting+youtube+link&total=1&ttsspeed=1&idx=0&client=tw-ob
2018-07-16 18:17:41,018 DEBUG gtts.tts status-0: 200
2018-07-16 18:17:41,019 DEBUG gtts.tts part-0 written to <_io.BufferedWriter name='/tmp/say.mp3'>
2018-07-16 18:17:41,019 DEBUG gtts.tts Saved to /tmp/say.mp3
2018-07-16 18:17:42,724 WARNING googleapiclient.discovery_cache file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth
Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/__init__.py", line 36, in autodetect
    from google.appengine.api import memcache
ImportError: No module named 'google.appengine'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/file_cache.py", line 33, in <module>
    from oauth2client.contrib.locked_file import LockedFile
ImportError: No module named 'oauth2client.contrib.locked_file'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/file_cache.py", line 37, in <module>
    from oauth2client.locked_file import LockedFile
ImportError: No module named 'oauth2client.locked_file'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/__init__.py", line 41, in autodetect
    from . import file_cache
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/file_cache.py", line 41, in <module>
    'file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth')
ImportError: file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth
2018-07-16 18:17:42,732 INFO googleapiclient.discovery URL being requested: GET https://www.googleapis.com/discovery/v1/apis/youtube/v3/rest
2018-07-16 18:17:43,411 INFO googleapiclient.discovery URL being requested: GET https://www.googleapis.com/youtube/v3/search?alt=json&key=AIzaSyDPM89CIlT8Honbi-14MTvn-Licx9GYQ7g&q=&part=id%2Csnippet

please help me with this

shivasiddharth, 2018-07-16:
Assistant recognizing your pronunciation of "stream" as "dream" is not an issue with the project. Change the keyword "stream" to something else that you are comfortable with. Just use find and replace.

closed 2018-07-16 by shivasiddharth

Shidiqadm, 2018-07-26:
S it's worked but as soon as I rebooted my device it automatically changes to default keyword stream