shivasiddharth/GassistPi #443: listen to the whole problem before closing the issue.

rohit366, 2018-07-17:


shivasiddharth, 2018-07-17:
If a problem has been marked closed and you think that you still have an issue there is a reopen button within the issue, click on that to reopen. This kind of unruly behavior is not tolerable. Learn to use the git wisely. I am sorry, I will no longer assist you for this behavior.

closed 2018-07-17 by shivasiddharth

rohit366, 2018-07-17:
thank for not assisting, I can's tolerate devs like you who just kill the spirit of newcomers, I understand that you might be having lots of work possibly heading google in your dreams but this is not the way to communicate your concern by telling me about my unruly behavior instead of listening to the problem first and giving your verdict just like that.