shivasiddharth/GassistPi #445: stream youtube

ibrahim5alil, 2018-07-22:
### Describe the bug:
```

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes/No

### 2. Is the issue related to assistant installation ?
Yes/No

## If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes/No

### 4. Is the assistant service restarting automatically?
Yes/No

## If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

### Attach the log file. Log file can be found in /tmp directory. Remeber to copy your log file before shutting down or restarting the Pi.

ibrahim5alil, 2018-07-22:
plz if anyone can help me 
i need to know what to edit and how in action.py to stream youtube

shivasiddharth, 2018-07-22:
Everything is given in detail in README. If you need further help for non-issue things, please use gitter.

closed 2018-07-22 by shivasiddharth

ibrahim5alil, 2018-07-22:
thank's for your reply sir, i will