shivasiddharth/GassistPi #449: alsa problem after new installation 23.07.18

outtacontrol216, 2018-07-23:
hi,

my mic and my speaker are 2 different USB devices. i set up a new rasbian installation and installed GassistPi from scratch. i used to disable the onboard soundcard by blacklisting. after that i enabled the USB devices to be default in /lib/modprobe.d/aliases.conf (outcomment options snd-usb-audio index=-2).
i installed the sound script for USB devices /scripts/install-usb-dac.sh. in the .asoundrc i had to edit the entry for the mic to:
pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:0,0"
        channels 1
    }
}
from:
pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:1,0"
        channels 1
    }
}
when i start GassistPi everything works fine. but i get lots of errors:
systemd[1]: Started Google Assistant-Voice.
python[1289]: [02a71e58] pulse audio output error: PulseAudio server connection failure: Connection refused
python[1289]: [02a61c98] pulse audio output error: PulseAudio server connection failure: Connection refused
python[1289]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
python[1289]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
python[1289]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
python[1289]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
python[1289]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
python[1289]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
python[1289]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
python[1289]: ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
python[1289]: ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
python[1289]: ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
python[1289]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
python[1289]: ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
python[1289]: ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
python[1289]: ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
python[1289]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
python[1289]: ALSA lib pcm_dsnoop.c:556:(snd_pcm_dsnoop_open) The dsnoop plugin supports only capture stream
python[1289]: ALSA lib pcm_dmix.c:990:(snd_pcm_dmix_open) The dmix plugin supports only playback stream
python[1289]: Cannot connect to server socket err = No such file or directory
python[1289]: Cannot connect to server request channel
python[1289]: jack server is not running or cannot be started
python[1289]: JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock

Can we fix that?

shivasiddharth, 2018-07-23:
These are not errors, just warnings. You tend to get that when using snowboy, there is nothing to be afraid of.

outtacontrol216, 2018-07-24:
ok fine thank you!

closed 2018-08-10 by shivasiddharth