shivasiddharth/GassistPi #450: hotword issue

serhatkimya, 2018-07-24:
thank you very much for added to hotword shivasiddharth.i have a problem, i tried three different hotwords and work perfectly but for example when listening to music, assistant choose the words and getting listening mode.it is so sensitive. i tried long wakeword but problem same.
thanks for everything.

shivasiddharth, 2018-07-24:
Sorry i am not able to understand.  You mean the custom hotwords are sensitive that even music is triggering them ? You can adjust the sentivity of the hotwords while creating them in snowboy or try changing the sensitivity here https://github.com/shivasiddharth/GassistPi/blob/e251b17f44bf82746cde2e373f26306017c5d6c7/src/main.py#L151

serhatkimya, 2018-07-25:
yes, you are right, i will try thanks.

serhatkimya, 2018-07-27:
i solved my problem by recording clear sound and everything is ok

closed 2018-07-27 by serhatkimya