shivasiddharth/GassistPi #452: "something went wrong, wait a few seconds" or "there is a glitch, try again in a few seconds"

pgs32, 2018-07-25:
Hi

I can manually start the assistant and it responds to the wake word, however when I ask a question it consistently replies _"something went wrong, wait a few seconds" or "there is a glitch, try again in a few seconds"

I have also included the last few lines from the terminal window showing a further main.py error

main.py: error: unrecognized arguments: --assistant-6459a-assistant-m4mdvn

I am not sure if the two issues are connected?

Do you have any suggestions please?

serhatkimya, 2018-07-25:
this situation most of time about bad internet connection.  you can try another internet connection like from phone tethering or cable.

feraco, 2018-07-25:
I have the same problem im not sure why. I have strong connection

shivasiddharth, 2018-07-27:
@pgs32 The something went wrong is definitely due to a slow server response speed (based on my experience). You Internet might be fast, but if Pi takes time to get a response from google's server, then  you tend to get that message. For an instance, even if i am doing some heavy downloading on the PC/Phone i get the message on Pi as a matter of fact even from Google Home. 
When we say slow internet, it does not mean your network speed, it means a slow response speed from google's server. Given the number of devices you have mentioned to be connected, i would not be surprised that you get this message.

shivasiddharth, 2018-07-27:
Duplicate of #140 .

closed 2018-07-27 by shivasiddharth

chiragatha, 2019-01-22:
Even I got the same issue.
Can you suggest some solution to it?