shivasiddharth/GassistPi #454: ask "what can you do?"

AmirCrimson, 2018-07-25:
### Describe the bug:
```
ask "what can you do?"
the assistant freezes this way
service stays stuck
but manually im able to break it and run it again.
can you plz fix this problem???
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes
### 2. Is the issue related to assistant installation ?
No

## If answer to question 2 is Yes, then paste the contents of the terminal below.

```
ON_RECOGNIZING_SPEECH_FINISHED:
{"text": "what can you do"}
ON_RENDER_RESPONSE:
{"text": "Here are some things you can ask for:", "type": 0}
ON_RENDER_RESPONSE:
{"text": "Here are some things you can ask for:", "type": 0}
ON_RENDER_RESPONSE:
{
"text": "Take a pick:\nTranslation · Sleep Timers · · Answers · Google Home · Calendar · Help · My Assistant · News · · Shopping · Sports · Timers · Weather · · Alarms · · · ",
"type": 0
}
ON_RENDER_RESPONSE:
{
"text": "Take a pick:\nTranslation · Sleep Timers · · Answers · Google Home · Calendar · Help · My Assistant · News · · Shopping · Sports · Timers · Weather · · Alarms · · · ",
"type": 0
}
ERROR:main:local variable 'URL' referenced before assignment
Traceback (most recent call last):
File "/home/pi/GassistPi/src/main.py", line 560, in 
Myassistant().main()
File "/home/pi/GassistPi/src/main.py", line 454, in main
feed(str(usrcmd).lower())
File "/home/pi/GassistPi/src/actions.py", line 308, in feed
feed=feedparser.parse(URL)
UnboundLocalError: local variable 'URL' referenced before assignment
```

## If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
No

## If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

### Attach the log file. Log file can be found in /tmp directory. Remeber to copy your log file before shutting down or restarting the Pi.

shivasiddharth, 2018-07-25:
Its nog a bug or an error. Once upon a time, google did not allow news streaming and the custom action that i implemented at that time is now interfering. Check [this](https://github.com/shivasiddharth/GassistPi/issues/346#issuecomment-391838204) out.

closed 2018-07-25 by shivasiddharth

AmirCrimson, 2018-07-25:
i have commented the line in main.py as you said and it stopped crashing 
thank you so very much 🥇