shivasiddharth/GassistPi #455: Wakeword problem

BarryAlle, 2018-07-26:
### Describe the bug:
```
Hi,Shiva, i have a problem about custom wakeword.When i add a Path of my model wakeworld in config.yaml,its return this:
ERROR:__main__:number of hotwords in decoder_model (4) and sensitivity (3) does not match
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 560, in <module>
    Myassistant().main()
  File "/home/pi/GassistPi/src/main.py", line 153, in __init__
    self.detector = snowboydecoder.HotwordDetector(models, sensitivity=self.sensitivity)
  File "/home/pi/GassistPi/src/snowboydecoder.py", line 100, in __init__
    "(%d) does not match" % (self.num_hotwords, len(sensitivity))
AssertionError: number of hotwords in decoder_model (4) and sensitivity (3) does not match

Please help me.
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes/No

### 2. Is the issue related to assistant installation ?
Yes/No

## If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes/No

### 4. Is the assistant service restarting automatically?
Yes/No

## If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

### Attach the log file. Log file can be found in /tmp directory. Remeber to copy your log file before shutting down or restarting the Pi.

shivasiddharth, 2018-07-26:
Can you please attach your config.yaml file?

shivasiddharth, 2018-07-26:
Right now, the assistant can take in max 2 wakewords. You have given 3.

BarryAlle, 2018-07-26:
but i gave it 2 wakewords still not working

shivasiddharth, 2018-07-26:
Please post the output of terminal when you use two wakewords.

BarryAlle, 2018-07-26:
My Terminal output:
ERROR:__main__:number of hotwords in decoder_model (3) and sensitivity (2) does not match
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 560, in <module>
    Myassistant().main()
  File "/home/pi/GassistPi/src/main.py", line 153, in __init__
    self.detector = snowboydecoder.HotwordDetector(models, sensitivity=self.sensitivity)
  File "/home/pi/GassistPi/src/snowboydecoder.py", line 100, in __init__
    "(%d) does not match" % (self.num_hotwords, len(sensitivity))
AssertionError: number of hotwords in decoder_model (3) and sensitivity (2) does not match

I think its still the same

BarryAlle, 2018-07-26:
Thats only work with alexa and snowboy but the other its not

shivasiddharth, 2018-07-26:
There is an issue with the jarvis.umdl hotword. Please refer to this https://github.com/Kitt-AI/snowboy/issues/447 one jarvis.umdl is being taken as two models.
If you want to use jarvis.umdl then make the following change in main.py in lines 151 and 152 in main.py

change `len(models)` to `(len(models)+1)` this is if you are using jarvis.umdl only.

BarryAlle, 2018-07-27:
How about i want to use 2 model of wakework?

BarryAlle, 2018-07-27:
i have train myself a model call friday but is not working when i use 2 model in config.yaml

shivasiddharth, 2018-07-27:
what are the two models you are using ?

BarryAlle, 2018-07-27:
nevermind, now its working perfectly.Thanks for spending your time to help me

shivasiddharth, 2018-07-27:
Closing this issue then.

closed 2018-07-27 by shivasiddharth