shivasiddharth/GassistPi #462: Raspi Pi0W no audio, and can't change language

fragolinux, 2018-07-29:
### Describe the bug:
```
raspberry pi 0 wifi, updated and run script as per instructions
no audio answers come out, even if audio is at working ("speaker-test -t wav" is just fine and volume, via alsamixer, is at maximum).
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No

### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
No

### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the contents of the terminal below
it starts normally, when i get those "PulseAudio server connection failure: Connection refused" the speaker is sounding the normal startup gassist sound, when i press the push button i get those STOPPED lines, if i try to press CTRL-C then, it never goes back to prompt and continues to make a blip sound every 10 seconds and writes those "WARNING:root:SoundDeviceStream read overflow (3200, 6400)" lines and button is full on and flashes every 30 seconds or so for a fraction of a second... i have to open a new terminal to reboot the device, when i get here...
```
pi@raspberrypi:~ $ /home/pi/env/bin/python -u /home/pi/GassistPi/src/pushbutton.py --project-id pi0w-aiy --device-model-id pi0w-aiy-speaker
[01a7c3e8] pulse audio output error: PulseAudio server connection failure: Connection refused
[01aae5c0] pulse audio output error: PulseAudio server connection failure: Connection refused
Stopped
stopping vlc
^C^CWARNING:root:SoundDeviceStream read overflow (3200, 6400)
^CWARNING:root:SoundDeviceStream read overflow (3200, 6400)
WARNING:root:SoundDeviceStream read overflow (3200, 6400)
WARNING:root:SoundDeviceStream read overflow (3200, 6400)
WARNING:root:SoundDeviceStream read overflow (3200, 6400)
WARNING:root:SoundDeviceStream read overflow (3200, 6400)
WARNING:root:SoundDeviceStream read overflow (3200, 6400)
WARNING:root:SoundDeviceStream read overflow (3200, 6400)
WARNING:root:SoundDeviceStream read overflow (3200, 6400)
WARNING:root:SoundDeviceStream read overflow (3200, 6400)
WARNING:root:SoundDeviceStream read overflow (3200, 6400)
WARNING:root:SoundDeviceStream read overflow (3200, 6400)
WARNING:root:SoundDeviceStream read overflow (3200, 6400)
WARNING:root:SoundDeviceStream read overflow (3200, 6400)
WARNING:root:SoundDeviceStream read overflow (3200, 6400)
WARNING:root:SoundDeviceStream read overflow (3200, 6400)
```

### Attach the log file. Log file can be found in /tmp directory. Remeber to copy your log file before shutting down or restarting the Pi.
no log files are present, just empty folders with random names...
```
drwxrwxrwt 16 root root 4096 Jul 29 12:00 .
drwxr-xr-x 22 root root 4096 Jun 27 01:09 ..
drwxrwxrwt  2 root root 4096 Jul 29 10:56 .font-unix
drwxrwxrwt  2 root root 4096 Jul 29 10:56 .ICE-unix
drwx------  2 pi   pi   4096 Jul 29 10:58 mc-pi
drwx------  2 root root 4096 Jul 29 12:00 mc-root
drwx------  2 pi   pi   4096 Jul 29 10:57 pulse-PKdhtXMmr18n
drwx------  3 root root 4096 Jul 29 10:56 systemd-private-7969bd266f1348ebb5ec657f3dc406e1-systemd-timesyncd.service-3EsmQ2
drwxrwxrwt  2 root root 4096 Jul 29 10:56 .Test-unix
drwx------  2 pi   pi   4096 Jul 29 11:03 tmp3uhx_4rc
drwx------  2 pi   pi   4096 Jul 29 11:39 tmpi7lehu1v
drwx------  2 pi   pi   4096 Jul 29 11:02 tmpoe_17918
drwx------  2 pi   pi   4096 Jul 29 11:58 tmpuryywqfi
drwx------  2 pi   pi   4096 Jul 29 10:57 tmpybe5zn61
drwxrwxrwt  2 root root 4096 Jul 29 10:56 .X11-unix
drwxrwxrwt  2 root root 4096 Jul 29 10:56 .XIM-unix
```

fragolinux, 2018-07-29:
oh, and i set language to italian in home app on android, but it says HI when it completes boot and starts services, and if i rerun the sample script, it understands italian words as they were in english and answers in english on console, but no audio from the speaker...

i run this for the sample script:
`/home/pi/env/bin/googlesamples-assistant-pushtotalk --project-id pi0w-aiy --device-model-id pi0w-aiy-speaker
`

shivasiddharth, 2018-07-29:
Please check this upstream issue https://github.com/googlesamples/assistant-sdk-python/issues/274 you need to use the display flag --display to authenticate when using pushtotalk for the first time after installation.

fragolinux, 2018-07-29:
ok, i'll take a look... but have i to redo all setup or there's a way to redo only the auth part? thanks for your excellent work :)

shivasiddharth, 2018-07-29:
Just stop the service and manually start the assistant, no need to run any scripts.

fragolinux, 2018-07-29:
but
ehm, and where i've to put that "--display" switch?
i just tried this (and installed pulseaudio as stated elsewhere, so no more warnings on startup...

sudo apt-get install pulseaudio
sudo systemctl stop gassistpi-push-button.service
sudo systemctl status gassistpi-push-button.service
/home/pi/env/bin/python -u /home/pi/GassistPi/src/pushbutton.py --project-id pi0w-aiy --device-model-id pi0w-aiy-speaker --display

it says HI, no more pulseaudio errors, but again i get the STOPPED errors and nothing else... and pressing ctrl-c restarts the blip sound...

Stopped
stopping vlc

shivasiddharth, 2018-07-29:
1. Are you using raspbian lite ? You need to use the raspbian desktop version for the --display flag to work.
2. I guess you are using AIY hat's pushbutton to trigger. The HAT's pushbutton has been configred to stop the music/media. (Please go through the README section of GPIOs used). 
3. I have added the custom wakeword functionality for Pi Zero as well, so GPIO usage for triggering has been disabled. To disable custom wakeword and enable GPIO triggering, please go through the custom wakeword triggering section of README.

To change the HAT's pushbutton as trigger button, the 23 in the following line to 22
https://github.com/shivasiddharth/GassistPi/blob/5c7236b36c128d2dd9482b28a6bc33e8e3b132c6/src/pushbutton.py#L215

and 22 in the following line to 23
https://github.com/shivasiddharth/GassistPi/blob/5c7236b36c128d2dd9482b28a6bc33e8e3b132c6/src/pushbutton.py#L835

fragolinux, 2018-07-29:
yes, i've raspbian lite, no desktop... i think i'll restart from scratch with display version and redo all setup, then ;)
yes, i'm using the aiy hat pushbutton connected to its own connector, don't know what gpio are those, i thought that selecting the right script during setup had taken care of selecting the right one for me...
i'll reread every readme you say then, and let you know how it goes :)
thanks for the help!

fragolinux, 2018-07-29:
i'm installing pixel desktop on top of lite install, so no need to redo everything again... but... this part is not very clear or is not updated to latest changelog, as you say that now pushbutton is disabled for pi zero and there's custom wakeword for it, too... but these lines seem to say that 1 line is for pi3 and the other for pi0, not both at the same time... can you clarify? thanks again

Enable the services - Pi3 and Armv7 users, enable the "gassistpi-ok-google.service" and Pi Zero, Pi A and Pi 1 B+ users, enable "gassistpi-push-button.service"
Previously a service was dedicated for stopbutton that stops music/radio etc. Now, its being run in a thread along with the assistant so you will not find the service.
sudo systemctl enable gassistpi-ok-google.service  
sudo systemctl enable gassistpi-push-button.service
Start the service - Pi3 and Armv7 users, start the "gassistpi-ok-google.service" and Pi Zero, Pi A and Pi 1 B+ users, start "gassistpi-push-button.service"
Previously a service was dedicated for stopbutton that stops music/radio etc. Now, its being run in a thread along with the assistant so you will not find the service.
sudo systemctl start gassistpi-ok-google.service  
sudo systemctl start gassistpi-push-button.service

fragolinux, 2018-07-29:
oh, you can add pulseaudio to requirements as it seems to have put away those warnings...

shivasiddharth, 2018-07-29:
> i'm installing pixel desktop on top of lite install, so no need to redo everything again... but... this part is not very clear or is not updated to latest changelog, as you say that now pushbutton [...]

1. Pi Zero uses grpc API and Pi 3 uses assistant library. grpc will work with Pi3 but not the other way around, so still we need to have separate scripts for Pi Zero and Pi3. To differentiate the scripts i have the aforementioned naming schema. The triggering by pushbutton has been disabled, but still you need a script to run the assistant on Pi Zero right. 

2. Not all audio devices give out quality audio with pulseaudio  and also, you may loose the ability to control volume from the bar on top if you install pulseaudio. I have already mentioned that those warnings need not be heeded. Those warnings have no effect on the project's functionality.

fragolinux, 2018-07-29:
1) understood, fine
2) as 1), removed pulseaudio
now, i run "/home/pi/env/bin/googlesamples-assistant-pushtotalk --project-id pi0w-aiy --device-model-id pi0w-aiy-speaker --display" in raspbian pixel, when it detects a phrase it opens chromium with a temporary html file, it asks for a couple of confirmation but it always end up with "an error occurred" and nothing else... i even logged in on google in chromium with same account used for all these tests...

fragolinux, 2018-07-29:
restarting from scratch, deleting project, starting from raspbian desktop and following your info...

shivasiddharth, 2018-10-07:
https://github.com/shivasiddharth/GassistPi/blob/642941a9ff87ed268b726ef13dbbf4f5b9b85300/src/main.py#L355

closed 2018-10-07 by shivasiddharth