shivasiddharth/GassistPi #466: Google Assistant will not start

WarmWinters, 2018-08-01:
### Describe the bug:

On a PI3 B+ I cloned  https://github.com/shivasiddharth/GassistPi and installed two days a ago.  I am using a USB Microphone and Bluetooth speakers via embedded Bluetooth in the Pi.  I have made several attempts to get this working.  Thanks for any assistance.

Google Assistant fully works with these commands. 
```
cd /home/pi
source env/bin/activate
python3 /home/pi/env/bin/google-assistant-demo
```

But will not work if setup to run as a service or if started manually with main.py I think I am missing something in the audio configuration.

```
This works.
aplay -D bluealsa:DEV=00:1F:81:94:70:C7 /usr/share/sounds/alsa/Front_Center.wav
```

```
As does this. Clip is recorded and played back.
arecord  -d 3 test.wav && aplay test.wav
```

```
No audio is heard with this however but I cannot see why :-(
speaker-test -t wav
speaker-test 1.1.3
Playback device is default
Stream parameters are 48000Hz, S16_LE, 1 channels
WAV file(s)
Rate set to 48000Hz (requested 48000Hz)
Buffer size range from 9600 to 4565228
Period size range from 480 to 4459
Using max buffer size 4565228
Periods = 4
........

```

Here are my files 

```
cat /home/pi/.asoundrc
pcm.!default {
  type asym
  capture.pcm "mic"
  playback.pcm "speaker"
}
pcm.mic {
  type plug
  slave {
    pcm "hw:1,0"
  }
}
pcm.speaker {
   type plug
        slave.pcm {
                type bluealsa
                device "00:1F:81:94:70:C7"
                profile "a2dp"
        }
}

cat /etc/asound.conf
pcm.!default {
  type asym
  capture.pcm "mic"
  playback.pcm "speaker"
}

[... 133 lines cut ...]
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
ALSA lib bluealsa-pcm.c:679:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA transport: No such device
ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
[FATAL:audio_input_stream.cc(47)] Input device could not be opened: default   <-------
Aborted`
```

### Attach the log file. Log file can be found in /tmp directory. Remeber to copy your log file before shutting down or restarting the Pi.

There is no log file.

shivasiddharth, 2018-08-01:
The issue here is the audio setup. I have enabled the custom wakeword option by default. So to simultaneously run the snowboy and google assistant, you need a dsnoop plugin. Which you dont have in your audio conf. So disable the custom wakeword in the config.yaml. After that you have the google assistant working normally. Go through the README for details/guidance to disable the custom wakeword.

WarmWinters, 2018-08-01:
Thank you.  I have disabled custom wakeword  and rebooted the pi but error remains. I will look into dsnoop as I would like to get custom wakeword working at some stage.

```
pi@computer:~ $ grep -i -a1 custom_wakeword   /home/pi/GassistPi/src/config.yaml
#To enable custom wakeword, change status to Enabled and restart the assistant.
Custom_wakeword:
  status: 'Disabled'

pi@computer:~ $ /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id 'femputer-e3f60-femputer-hwz8p2'
[01790da0] pulse audio output error: PulseAudio server connection failure: Connection refused
[0179bf20] pulse audio output error: PulseAudio server connection failure: Connection refused
ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
ALSA lib bluealsa-pcm.c:679:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA transport: No such device
ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
[FATAL:audio_input_stream.cc(47)] Input device could not be opened: default 
Aborted

```

WarmWinters, 2018-08-01:
You are correct Sir.  Its working now.  I will research dsnoop.  Many thanks for creating this project and your help!

shivasiddharth, 2018-08-02:
Hello, I would like to give users as many options as possible and freedom to use a number of different setups. I would like to add the bluetooth audio setup to the project. Can you kindly mention the bluetooth device that you are using and the steps that you went through to set it up? I shall try to investigate and add dsnoop to bluetooth setups.

WarmWinters, 2018-08-02:
Hello.  No problem. I will do that as soon as I have some time.

WarmWinters, 2018-08-04:
I am using an HMDX Bluetooth speaker.   These commands were ran on a fresh install of Stretch. No need to install additional packages.

``` 
pi@raspberrypi:~ $ bluetoothctl
[NEW] Controller B8:27:EB:41:F3:97 raspberrypi [default]
[NEW] Device 00:1F:81:94:70:C7 HMDX RAVE
[CHG] Device 00:1F:81:94:70:C7 Connected: no

[bluetooth]# agent on
Agent registered

[bluetooth]# trust 00:1F:81:94:70:C7
[CHG] Device 00:1F:81:94:70:C7 Trusted: yes
Changing 00:1F:81:94:70:C7 trust succeeded

[bluetooth]# pair 00:1F:81:94:70:C7
Attempting to pair with 00:1F:81:94:70:C7
[CHG] Device 00:1F:81:94:70:C7 Connected: yes
Request PIN code
[agent] Enter PIN code: 0000
[CHG] Device 00:1F:81:94:70:C7 Paired: yes
Pairing successful
[CHG] Device 00:1F:81:94:70:C7 Connected: no

[bluetooth]# connect 00:1F:81:94:70:C7
Attempting to connect to 00:1F:81:94:70:C7

[HMDX RAVE]# info 00:1F:81:94:70:C7
Device 00:1F:81:94:70:C7
        Name: HMDX RAVE
        Alias: HMDX RAVE
        Class: 0x240414
        Icon: audio-card
        Paired: yes
        Trusted: yes
        Blocked: no
        Connected: yes
        LegacyPairing: yes
        UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)

[HMDX RAVE]# exit

aplay -D bluealsa:DEV=00:1F:81:94:70:C7 /usr/share/sounds/alsa/Front_Center.wav
Playing WAVE '/usr/share/sounds/alsa/Front_Center.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono

```

contactwajeeh, 2018-08-09:
I was able to setup GassistPi with nowakeword main.py file, but only assistant works with that, no youtube playback, Gmusic etc doesnt work with this  setup.

oceank82, 2018-09-02:
Hello!

I have the Assistantspi running well with both assistants. How can I make them using my bluetooth speaker? I cannot get it to work, anyone?

Hoeby, 2018-10-28:
Sorry for adding to this old thread, but i noticed a few people have problems getting the bluetooth working. I got it working and would to share it with all. This works whit dsnoop active, so also custom wake-word should work.
I am using a PS3 eye USB microphone and a havit M16 bluetooth speaker

Step 1:
Install GassistPi like you wanted to use the jack plug as audio output.
This was for me the start of the install. 

Step 2: 
Follow the steps to enable and pair the bluetooth speaker on the raspberry pi. I am using a 3b+, don't know if these steps works on other pi's. This has nothing to do with GassistPi.
Follow the instruction on the next link: https://sigmdel.ca/michel/ha/rpi/bluetooth_01_en.html
I had the problem that the PI didn't autoconnect (pair bluetooth) after reboot. I made a script for it and put it in autodesk, to start it with power-on. 

Step 3:
Change your .asoundrc file to (make sure to make a copy or printscreen, just in case)
There where i written "place your bluetooth mac ID here", only change the text to mac ID, but keep the " and " in the script

```
pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave { 
        pcm "hw:1,0"
        channels 1
    }
}

pcm.bluetooth {
    type plug 
    slave.pcm {
        type bluealsa
        device "place your bluetooth mac ID here"
        profile "a2dp"
    }
}

pcm.!default {
    type asym
    playback.pcm {
        type plug
        slave.pcm "bluetooth"
    }
    capture.pcm {
        type plug
        slave.pcm "dsnooper"
    }
}
```

bachoo786, 2019-03-01:
@Hoeby I followed your method however when I test my sound it doesnt work.

what I noticed was that the .asoundrc file changes back to default with the default bluetooth settings when I restart my rpi3b+

this is what the .asoundrc file becomes after I restart:

```
pcm.!default {
        type plug
        slave.pcm {
                type bluealsa
                device "20:18:08:30:A9:66"
                profile "a2dp"
        }
}

ctl.!default {
        type bluealsa
}
```

Hoeby, 2019-03-02:
your .asoundrc file doesn't have the right permissions. that is why it changes back.
Do a chmod 777 on your .asoundrc file and try again after you did that.

bachoo786, 2019-03-02:
Which file do I need to edit ? Is this the one in the home folder ?

bachoo786, 2019-03-02:
@hoeby do I need to follow the link that you sent because the latest raspbian version works with Bluetooth already and it connects to my Bluetooth speaker whenever I restart my pi .

I get errors with my mic saying that it is not connected but I'm using the USB mic and it doesn't work . The mic is working and I can see that the USB connection is working .

bachoo786, 2019-03-02:
this is my log when I manually start it:

```
INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
INFO:gmusicapi.Mobileclient1:version: 12.0.0
INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
INFO:gmusicapi.Mobileclient1:initialized
INFO:gmusicapi.Mobileclient1:logged out
/home/pi/env/lib/python3.5/site-packages/oauth2client/_helpers.py:255: UserWarning: Cannot access /home/pi/.local/share/gmusicapi/mobileclient.cred: No such file or directory
  warnings.warn(_MISSING_FILE_MESSAGE.format(filename))
2019-03-02 11:45:22,700 - gmusicapi.Mobileclient1 (shared:183) [WARNING]: could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
WARNING:gmusicapi.Mobileclient1:could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
INFO:gmusicapi.Mobileclient1:failed to authenticate
[03131580] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[03138f28] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
USB-MIC-JACK
Initializing GPIOs 5 and 6 for assistant activity indication
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.front.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM front
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround40.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround40
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround41
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround50
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround51
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround71.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround71
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM iec958
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
[... cut ...]
  File "/home/pi/env/lib/python3.5/site-packages/pyaudio.py", line 750, in open
    stream = Stream(self, *args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/pyaudio.py", line 441, in __init__
    self._stream = pa.open(**arguments)
OSError: [Errno -9997] Invalid sample rate

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 1026, in <module>
    logger.exception(error)
NameError: name 'logger' is not defined

```

@shivasiddharth can you please help me?

Hoeby, 2019-03-02:
> @Hoeby do I need to follow the link that you sent because the latest raspbian version works with Bluetooth already and it connects to my Bluetooth speaker whenever I restart my pi .
> I get errors [...]

The link has nothing to do with the problem you have about changing .asoundrc after reboot.

Log in on your Pi and give the next command
`sudo chmod 0444 ./.asoundrc `
There should be no error, when there is an error, please post the error here in this thread.
After done the command, change the .asoundrc to what you need and reboot the Pi. 
After reboot, check to .asoundrc file if it's OK by now.

You first need to get the problems solved with your .asoundrc file, before going further.

First test this, but be sure that the .asoundrc file doesn't change after reboot.  This needs to be stable. When it is stable, but still having errors. Then we can look further.