shivasiddharth/GassistPi #468: Audio Settings with Respeaker 4 Mic Hat.

LordTyrion123, 2018-08-04:
### Describe the bug:
```
Hi @shivasiddharth. Thanks for all the hard work in this project. A quick question though. I'm using a Respeaker 4 Mic Hat as the microphone. The Respeaker is attatched to the GPIO pins of the Raspberry PI 3.  I tried both **CUSTOM VOICE HAT users** and *USB DAC or USB Sound CARD users* but it doesn't seem to recognize the Respeaker. I tested the installation on a normal USB microphone and it works perfectly. Is there any way to configure it to the Respeaker?
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
Yes

## If answer to question 2 is Yes, then paste the contents of the terminal below.

```
[01dd1448] pulse audio output error: PulseAudio server connection failure: Connection refused
[01dbf738] pulse audio output error: PulseAudio server connection failure: Connection refused
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
ALSA lib pcm_dmix.c:990:(snd_pcm_dmix_open) The dmix plugin supports only playback stream
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
[FATAL:audio_input_stream.cc(47)] Input device could not be opened: default
Aborted
```

shivasiddharth, 2018-08-04:
You need to setup the respeaker device first. Follow their guidelines. Not all audio config files work with all devices, so randomly picking up config for your respeaker will not work. Even after setting up respeaker you might run into issues, if you dont have dsnoop plugin enabled. Follow #466

LordTyrion123, 2018-08-06:
@shivasiddharth, I've installed my respeaker and recording works perfectly. Do I now add the dsnoop to the asoundrc and asound.conf?

LordTyrion123, 2018-08-06:
@shivasiddharth Everything is working now. Had to add dsnoop to my asound.conf and asoundrc files and it worked like a charm. Thanks for the help!

closed 2018-08-06 by LordTyrion123

shivasiddharth, 2018-08-08:
I will include Respeaker support in the next update, so that everything gets setup in a single go. Can you share your modified asound.conf  file. That will save some time for me.

shivasiddharth, 2018-09-01:
@LordTyrion123 Am in the process of incorporating support for respeaker hat. Can you please share the audio config files that you are using.

LordTyrion123, 2018-09-03:
Yeah. No problem!

**asoundrc File**

```
pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:0,1"
        channels 1
    }
}

    }
pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "hw:0,0"
        }
        capture.pcm {
                type plug
                slave.pcm "dsnooper"
        }
}
```
**asound.conf File**

```
pcm.!default {
    type asym
    playback.pcm {
        type plug
        slave.pcm "hw:ALSA"
    }
    
    capture.pcm {
	slave.pcm "dsnooper"
    }
}

pcm.dsnooper {
    #type plug
    #slave.pcm "hw:seeed4micvoicec"
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:seeed4micvoicec"
        channels 4
    }
}
```

shivasiddharth, 2018-09-03:
I have made the Respeaker indicators to work with this project. Give it a shot. Uninstall the previous Respeaker installation and use 3.6 in README.