shivasiddharth/GassistPi #469: error spotify

aandroide, 2018-08-05:
### Describe the bug:
```
spotify don't work

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No

## If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
No

## If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```
ERROR:__main__:local variable 'playlist_name' referenced before assignment
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 572, in <module>
    Myassistant().main()
  File "/home/pi/GassistPi/src/main.py", line 565, in main
    spotify_playlist_select(str(usrcmd).lower())
  File "/home/pi/GassistPi/src/actions.py", line 1311, in spotify_playlist_select
    playlists,num=scan_spotify_playlists()
  File "/home/pi/GassistPi/src/actions.py", line 1287, in scan_spotify_playlists
    playlistdetails.append(playlist_name)
UnboundLocalError: local variable 'playlist_name' referenced before assignment

```

### Paste the contents of the terminal below
```

```

### Attach the log file. Log file can be found in /tmp directory. Remeber to copy your log file before shutting down or restarting the Pi.

moreover the log file is not created

shivasiddharth, 2018-08-05:
can you uncomment the following line and see if it prints. https://github.com/shivasiddharth/GassistPi/blob/faace9d8af3881f3f41a49134ce707ad0454379a/src/actions.py#L1276

aandroide, 2018-08-05:
no, 
it does not print anything

shivasiddharth, 2018-08-05:
can you check if this is printing anything https://github.com/shivasiddharth/GassistPi/blob/faace9d8af3881f3f41a49134ce707ad0454379a/src/actions.py#L1271

aandroide, 2018-08-05:
{'Playlists': [0, 0, 0, 0]}

shivasiddharth, 2018-08-05:
Do you have 4 playlists? can you double check if the "music" playlist is public?

aandroide, 2018-08-05:
yes I have 4 playlists and they are all public

aandroide, 2018-08-05:
I discovered the problem. yesterday I added 2 playlists not created by me, but by a friend, who sent me via whatsapp, then I added it to my library.
now I removed them and it worked.
Is this a spotipy bug?

shivasiddharth, 2018-08-05:
Its not exactly a bug. Its well known that playlists have to be public. Let me do one thing, I will leave this open and add an enhancement label. I will change the codes a bit to check the ownership of the playlists that people have and make it not crash if it has 3rd party playlists. For testing purpose, can you share your playlist?

aandroide, 2018-08-05:
yes, is this: https://open.spotify.com/user/31fy67sgdqj3zrlmm2rjgamgckh4/playlist/7ofQzt0d3apbnMzZm1t4Zm?si=YbkC-xb8QM2UeIggiCA-zg&nd=1

aandroide, 2018-08-07:
I have updated the project # 460
but I get this error:

ERROR:__main__:too many values to unpack (expected 2)
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 681, in <module>
    Myassistant().main()
  File "/home/pi/GassistPi/src/main.py", line 674, in main
    spotify_playlist_select(str(usrcmd).lower())
  File "/home/pi/GassistPi/src/actions.py", line 1410, in spotify_playlist_select
    vlcplayer.spotify_player(currenttrackid)
  File "/home/pi/GassistPi/src/mediaplayer.py", line 132, in spotify_player
    fullurl,urlid=youtube_search(tracks[trackid])
ValueError: too many values to unpack (expected 2)

shivasiddharth, 2018-08-07:
#460 is crashing spotify, i have reverted the #460 back and updated the master.

aandroide, 2018-08-07:
I confirm that it now works by restoring the master, so that commit is to be reviewed

shivasiddharth, 2018-11-29:
I think i have fixed the third party playlist issue. Can you test and confirm?

closed 2018-11-29 by shivasiddharth

shivasiddharth, 2018-12-01:
So you are saying that, you are not able to play songs from the third party playlist ?

aandroide, 2018-12-01:
Yes, exactly

commit d9139ef1 (2018-11-29): Update actions.py

Fixes #469