shivasiddharth/GassistPi #473: Service has no audio

ThatThinggyMcBobber, 2018-08-08:
### Describe the bug:
```Working fine manually but auto start doesn't have any audio. The service is installed and running. Here is the output of status.
 
Linux raspberrypi 4.14.52-v7+ #1123 SMP Wed Jun 27 17:35:49 BST 2018 armv7l

The programs included with the Debian GNU/Linux system are free software;
* gassistpi-ok-google.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi-ok-google.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2018-08-08 11:03:47 CDT; 29min ago
 Main PID: 2804 (python)
   CGroup: /system.slice/gassistpi-ok-google.service
           |-2804 /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id ***************************
           `-2816 /usr/bin/pulseaudio --start --log-target=syslog

Aug 08 11:03:54 raspberrypi python[2804]: jack server is not running or cannot be started
Aug 08 11:03:54 raspberrypi python[2804]: JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Aug 08 11:03:55 raspberrypi python[2804]: device_model_id: ******************************
Aug 08 11:03:55 raspberrypi python[2804]: device_id: *****************************************
Aug 08 11:03:56 raspberrypi python[2804]: ON_MUTED_CHANGED:
Aug 08 11:03:56 raspberrypi python[2804]:   {"is_muted": false}
Aug 08 11:03:56 raspberrypi python[2804]: ON_MUTED_CHANGED:
Aug 08 11:03:56 raspberrypi python[2804]:   {"is_muted": false}
Aug 08 11:03:56 raspberrypi python[2804]: ON_START_FINISHED
Aug 08 11:03:56 raspberrypi python[2804]: ON_START_FINISHED
~

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes/No
 YES

### 2. Is the issue related to assistant installation ?
Yes/No
No

## If answer to question 2 is Yes, then paste the contents of the terminal below.

## If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes/No
yes, but not the service part.

### 4. Is the assistant service restarting automatically?
Yes/No
no

## If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```
pi@raspberrypi:~ $ /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id '*******************************'                             
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
ALSA lib pcm_dmix.c:990:(snd_pcm_dmix_open) The dmix plugin supports only playback stream
ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
ALSA lib pcm_dsnoop.c:556:(snd_pcm_dsnoop_open) The dsnoop plugin supports only capture stream
ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
device_model_id: ***************************
device_id: ****************************

ON_MUTED_CHANGED:
  {"is_muted": false}
ON_MUTED_CHANGED:
  {"is_muted": false}
ON_START_FINISHED
ON_START_FINISHED
```

### Attach the log file. Log file can be found in /tmp directory. Remeber to copy your log file before shutting down or restarting the Pi.

shivasiddharth, 2018-08-08:
Looks like a duplicate of #465. Can you confirm the audio scripts that you used and the audio setup that you have.

shivasiddharth, 2018-08-13:
By default, the Assistant's volume will be barely audible. Did you try giving the set volume to maximum command as mentioned in the readme?

ThatThinggyMcBobber, 2018-08-13:
it works great if manual starts but wont auto start. the audio is AIY-HAT and we are using the with google voice kit.

shivasiddharth, 2018-08-28:
Check the model-id that you have added in the systemd file. The issue is a duplicate of #465. This is an upstream bug.

closed 2018-08-28 by shivasiddharth