shivasiddharth/GassistPi #475: Google Assistant's Voice Not Playing

LordTyrion123, 2018-08-13:
### Describe the bug:
```
The Google assistant boots up fine, but when I ask it a general question, such as the time, no voice comes out. The render response prints perfectly as well, but the assistant does not say or speak that response.
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No

## If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
No

## If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```
The command did not crash the assistant. But the command given was What's the time?
```

### Paste the contents of the terminal below
```
ON_CONVERSATION_TURN_STARTED

ON_END_OF_UTTERANCE

ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "what's the time"}
{'text': "what's the time"}
ON_RENDER_RESPONSE:
  {"text": "It's 11:38.", "type": 0}
{'text': "It's 11:38.", 'type': <RenderResponseType.TEXT: 0>}
ON_RESPONDING_STARTED:
  {"is_error_response": false}

ON_RESPONDING_FINISHED

ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}

```

shivasiddharth, 2018-08-13:
Duplicate of #473

LordTyrion123, 2018-08-13:
@shivasiddharth . I don't think it's a duplicate of #473. In the mentioned issue, he says it works fine in manual, whereas here I am not getting the Google Assistant's voice itself. Additionally, #473 links to #465 in which long HTML is being printed. The only problem I am facing is the absence of the Google Assistant's voice.

shivasiddharth, 2018-08-13:
By default, the Assistant's volume will be barely audible. Did you try giving the set volume to maximum command as mentioned in the readme?

LordTyrion123, 2018-08-13:
I guess that was the mistake. I must have forgot that the Assistant has a different volume compared to the other Sound files. Thanks for the help! But is there any way to set the Assistant's voice at maximum in the code so it happens automatically?

shivasiddharth, 2018-08-28:
Closing this for inactivity. There is no way to auto set the volume, it has to be done manually.

closed 2018-08-28 by shivasiddharth