shivasiddharth/GassistPi #481: Stuck at "Playing assistant response"

jbrepogmailcom, 2018-08-15:
### Describe the bug:
```
After installation, when script starts test, it listens for command, receives command, prints "Playing assistant response" and then gets stuck. I have USB Mic and USB sound card (in one device). I have seen this error few months ago and it got fixed by manual reconfiguration of sound devices. Could you please instruct me how to check what are the USB devices and in which config file to put proper device IDs? Thanks.
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
Yes

## If answer to question 2 is Yes, then paste the contents of the terminal below.

```
Press Enter to send a new request...
INFO:root:Recording audio request.
WARNING:root:SoundDeviceStream read overflow (3200, 6400)
INFO:root:End of audio request detected.
INFO:root:Stopping recording.
INFO:root:Finished playing assistant response.
Press Enter to send a new request...
INFO:root:Recording audio request.
INFO:root:Transcript of user request: "hello".
INFO:root:Transcript of user request: "how long".
INFO:root:Transcript of user request: "how long".
INFO:root:End of audio request detected.
INFO:root:Stopping recording.
INFO:root:Transcript of user request: "hello".
INFO:root:Playing assistant response.
```

## If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes/No

### 4. Is the assistant service restarting automatically?
Yes/No

## If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

### Attach the log file. Log file can be found in /tmp directory. Remeber to copy your log file before shutting down or restarting the Pi.

shivasiddharth, 2018-08-15:
Since the installation is successful i think i can close #480. 
First up, this issue is not related to installation. Issues with installation means abrupt crashing of the installer or unsuccessful installation attempt.  Secondly, there is hardly any details provided here.
1. Did you start from scratch or just tried installing over previous assistant installation? 
2. Whether the Raspbian is Desktop version or Lite? 
3. Did you try running the assistant manually ?

 Want of general help for checking USB ids etc falls outside the scope of issues. Time and again, i advise people to use the gitter for non-issue posts.

jbrepogmailcom, 2018-08-16:
I tried both. To install over installation that I had from the past. Since it did not work I have created completely new SD card with Raspbian Lite and ran installation script. Again it got stuck at Playing response. My IDs for USB devices are 0,1 and 0,1. I will try to start manually.

jbrepogmailcom, 2018-08-19:
Hello. Tried run manually main.py, got error no module named kodijson. Launched environment in /home/pi/env/bin/activate, but same error.

Launched pushbutton.py, but got no response for several minutes. Did not instruct me if and what parameters should I add. I even could not Ctrl-C, I could only Ctrl-Z.

Can you help please? Would it be possible to make image of working PiZeroW installation, where I can add my own configuration json file and launch with parameters?

Thank you very much, Jan

jbrepogmailcom, 2018-08-22:
Could this be caused by incorrect audio configuration? Several months ago I could have fixed it by manually configuring audio, but I do not remember now

shivasiddharth, 2018-08-23:
Recently folks at google made a change to the audio helpers for grpc (https://github.com/googlesamples/assistant-sdk-python/commit/d05f675f5f28895efac6245a2f129e83adfd45da) , may be that could have impacted.  I will test it over the weekend.

jbrepogmailcom, 2018-09-07:
Hello, dont you have any news please?

Thanks, Jan

jbrepogmailcom, 2019-07-20:
Could not be repeated

closed 2019-07-20 by jbrepogmailcom