shivasiddharth/GassistPi #491: THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE

jbrepogmailcom, 2018-08-21:
### Describe the bug:
```
Clean install on Raspbian Desktop. Got this error:

THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE. If you have updated the package versions, please update the hashes. Otherwise, examine the package contents carefully; someone may have tampered with them.
    grpcio==1.14.1 from https://www.piwheels.org/simple/grpcio/grpcio-1.14.1-cp35-cp35m-linux_armv6l.whl#sha256=a4e46be841646a97a600c71d77f7b7701ad0ee11fffdeeb51f5fe3454f926812:
        Expected sha256 a4e46be841646a97a600c71d77f7b7701ad0ee11fffdeeb51f5fe3454f926812
             Got        57c1b34424c51c385a982a4faf8f9d97f30102ff1b0111971b0b5c1bd4b182fe

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes/No

### 2. Is the issue related to assistant installation ?
Yes/No

## If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes/No

### 4. Is the assistant service restarting automatically?
Yes/No

## If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

### Attach the log file. Log file can be found in /tmp directory. Remeber to copy your log file before shutting down or restarting the Pi.

Vino-git, 2018-08-21:
Requirement already satisfied: pbr>=0.11 in ./env/lib/python3.5/site-packages (from mock>=0.7.0->gmusicapi>=11.0.1->-r /home/pi/GassistPi/Requirements/GassistPi-pip-requirements.txt (line 6)) (4.2.0)
Requirement already satisfied: beautifulsoup4 in ./env/lib/python3.5/site-packages (from MechanicalSoup>=0.4.0->gmusicapi>=11.0.1->-r /home/pi/GassistPi/Requirements/GassistPi-pip-requirements.txt (line 6)) (4.6.3)
Requirement already satisfied: lxml in ./env/lib/python3.5/site-packages (from MechanicalSoup>=0.4.0->gmusicapi>=11.0.1->-r /home/pi/GassistPi/Requirements/GassistPi-pip-requirements.txt (line 6)) (4.2.4)
Requirement already satisfied: netifaces!=0.10.5 in ./env/lib/python3.5/site-packages (from zeroconf>=0.17.7->PyChromecast>=2.2.0->-r /home/pi/GassistPi/Requirements/GassistPi-pip-requirements.txt (line 11)) (0.10.7)
Requirement already satisfied: cachetools>=2.0.0 in ./env/lib/python3.5/site-packages (from google-auth>=1.4.1->google-api-python-client>=1.6.7->-r /home/pi/GassistPi/Requirements/GassistPi-pip-requirements.txt (line 15)) (2.1.0)
Building wheels for collected packages: spotipy
  Running setup.py bdist_wheel for spotipy ... done
  Stored in directory: /tmp/pip-ephem-wheel-cache-cjsdct46/wheels/98/31/d8/02e824694a8c8e6f3451008196487ba388f623657b47c04a93
Successfully built spotipy
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting grpcio==1.14.1 from https://www.piwheels.org/simple/grpcio/grpcio-1.14.1-cp35-cp35m-linux_armv6l.whl#sha256=a4e46be841646a97a600c71d77f7b7701ad0ee11fffdeeb51f5fe3454f926812
  Using cached https://www.piwheels.org/simple/grpcio/grpcio-1.14.1-cp35-cp35m-linux_armv6l.whl
THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE. If you have updated the package versions, please update the hashes. Otherwise, examine the package contents carefully; someone may have tampered with them.
    grpcio==1.14.1 from https://www.piwheels.org/simple/grpcio/grpcio-1.14.1-cp35-cp35m-linux_armv6l.whl#sha256=a4e46be841646a97a600c71d77f7b7701ad0ee11fffdeeb51f5fe3454f926812:
        Expected sha256 a4e46be841646a97a600c71d77f7b7701ad0ee11fffdeeb51f5fe3454f926812
             Got        57c1b34424c51c385a982a4faf8f9d97f30102ff1b0111971b0b5c1bd4b182fe

jbrepogmailcom, 2018-08-22:
Fixed by editing install script and replacing SHA256 hash with the correct one

closed 2018-08-22 by jbrepogmailcom

se12341, 2019-05-02:
pip install tensorflow
Collecting tensorflow
  Downloading https://www.piwheels.org/simple/tensorflow/tensorflow-1.13.1-cp27-none-linux_armv7l.whl (92.5MB)
    43% |█████████████▊                  | 39.8MB 273kB/s eta 0:03:13
THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE. If you have updated the package versions, please update the hashes. Otherwise, examine the package contents carefully; someone may have tampered with them.
    tensorflow from https://www.piwheels.org/simple/tensorflow/tensorflow-1.13.1-cp27-none-linux_armv7l.whl#sha256=889f4ed0615dc7ebba7781b7f68f83c7d5d6997e1233f8beb22bd6e478fdbdb1:
        Expected sha256 889f4ed0615dc7ebba7781b7f68f83c7d5d6997e1233f8beb22bd6e478fdbdb1
             Got        d4f2538b899d90a87fa87aed7cdf1898f124bf478aca692b90cdf8d03a84cc99
****

TheMasterRoot, 2019-05-07:
same problem here

jainthedestroyer, 2019-06-27:
same problem...

huwade, 2019-07-11:
same problem...

tertianalpaca01, 2019-07-22:
I too got the same issue

sdevane, 2019-09-22:
whats the solution to this please?

commit 18c92508 (2018-08-22): Update gassist-installer-pi-zero.sh

Fixes #491