shivasiddharth/GassistPi #493: Illegal instruction / Pi Zero is back

jbrepogmailcom, 2018-08-22:
### Describe the bug:
```
Installation ends with Error Illegal instruction. Tried various combinations of env and python -u, does not work. Clean install on Raspbian 9. Had to fix SHA256 in install script

TERMINAL:
credentials saved: /home/pi/.config/google-oauthlib-tool/credentials.json
Testing the installed google assistant. Make a note of the generated Device-Id
/home/pi/GassistPi/scripts/gassist-installer-pi-zero.sh: line 60:  5142 Illegal instruction     googlesamples-assistant-pushtotalk --project-id $projid --device-model-id $modelid
pi@raspberrypi:~ $ /home/pi/env/bin/python -u /home/pi/GassistPi/src/pushbutton.py --project-id r2d2-89647 --device-model-id r2d2-89647-r2d2-fb4ex9
Illegal instruction
pi@raspberrypi:~ $ /home/pi/env/bin/python /home/pi/GassistPi/src/pushbutton.py --project-id r2d2-89647 --device-model-id r2d2-89647-r2d2-fb4ex9
Illegal instruction
pi@raspberrypi:~ $ source env/bin/activate
(env) pi@raspberrypi:~ $ /home/pi/env/bin/python /home/pi/GassistPi/src/pushbutton.py --project-id r2d2-89647 --device-model-id r2d2-89647-r2d2-fb4ex9
Illegal instruction
(env) pi@raspberrypi:~ $ /home/pi/env/bin/python -u /home/pi/GassistPi/src/pushbutton.py --project-id r2d2-89647 --device-model-id r2d2-89647-r2d2-fb4ex9
Illegal instruction
(env) pi@raspberrypi:~ $ 

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
Yes

## If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
No

### 4. Is the assistant service restarting automatically?
Yes/No

## If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

### Attach the log file. Log file can be found in /tmp directory. Remeber to copy your log file before shutting down or restarting the Pi.

shivasiddharth, 2018-08-22:
The illegal instruction is because of wrong grpcio package. I have fixed the installer script. Can you run that again.

jbrepogmailcom, 2018-08-22:
Hello! I wiped GassistPi, cloned from github again and ran installer again. Infortunately, I got the same error:

credentials saved: /home/pi/.config/google-oauthlib-tool/credentials.json
Testing the installed google assistant. Make a note of the generated Device-Id
/home/pi/GassistPi/scripts/gassist-installer-pi-zero.sh: line 60: 13988 Illegal instruction     googlesamples-assistant-pushtotalk --project-id $projid --device-model-id $modelid
pi@raspberrypi:~ $

shivasiddharth, 2018-08-22:
Can you try from scratch again ? I have made changes to the installer script.

closed 2018-08-22 by shivasiddharth

jbrepogmailcom, 2018-08-22:
Sorry, still the same error. Should I open you my RPi Zero for remote ssh access?

credentials saved: /home/pi/.config/google-oauthlib-tool/credentials.json
Testing the installed google assistant. Make a note of the generated Device-Id
/home/pi/GassistPi/scripts/gassist-installer-pi-zero.sh: line 60: 15117 Illegal instruction     googlesamples-assistant-pushtotalk --project-id $projid --device-model-id $modelid
pi@raspberrypi:~ $

shivasiddharth, 2018-08-22:
Did you erase the card and try? Coz just re-running the installer will not work as you already have some version of grpcio installed.

jbrepogmailcom, 2018-08-22:
Hello! I have done completely clean install and finally got same problem as with clean install on Raspbian Lite. The test at end of installation script gets stuck at playing the response. Can not be killed by Ctrl-C, only suspended by Ctrl-Z. So I think you can close this problem, but you should keep Issue #481 open.

INFO:root:Transcript of user request: "hello Google".
INFO:root:Playing assistant response.
^C
^Z
[1]+  Stopped                 sudo /home/pi/GassistPi/scripts/gassist-installer-pi-zero.sh
pi@raspberrypi:~ $

jbrepogmailcom, 2019-07-19:
Went back from Buster to Stretch version 2018-11 and got the Illegal instruction again:

credentials saved: /home/pi/.config/google-oauthlib-tool/credentials.json
Testing the installed google assistant. Make a note of the generated Device-Id
./GassistPi/scripts/gassist-installer.sh: line 225: 12520 Illegal instruction     googlesamples-assistant-pushtotalk --project-id $projid --device-model-id $modelid

commit 9cbdbf09 (2018-08-22): Update gassist-installer-pi-zero.sh

Fixes #493