shivasiddharth/GassistPi #494: GPIO duplicates

georgefst, 2018-08-22:
GassistPi uses some of the same GPIOs as my own project. When I run both simultaneously, the Pi shuts down. Due to the physical setup it would be difficult for me to change some of these to use other pins.

Is there any way to disable the GPIO functionality? Or would it be possible for you to change GassistPi to use a config file to set the pins used for each feature (as I can see others having similar problems in future)?

Otherwise my only option is seemingly to fork GassistPi and change the hardcoded values, which seems like overkill.

shivasiddharth, 2018-08-22:
The config.yaml lists few of the GPIOs. In the next update, i shall list all the GPIOs there.

georgefst, 2018-08-22:
Great, thanks

shivasiddharth, 2018-08-23:
Hi, I have made changes to the project, listing all GPIOs in the config.yaml file. Please replace your existing main.py, actions.py, config.yaml and pushbutton.py files with the respective files from this branch (https://github.com/shivasiddharth/GassistPi/tree/disable-ok-google). Give it a run and let me know as to how it goes. This has not been merged with the master yet, there are quite few other things that i am working on so once all that is done, i shall merge it with the master.

georgefst, 2018-08-23:
Done exactly that, and changed some of the GPIO numbers. When run it, I get the following errors:
`[015f4ff8] pulse audio output error: PulseAudio server connection failure: Connection refused
[0162da68] pulse audio output error: PulseAudio server connection failure: Connection refused
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 35, in <module>
    from actions import say
  File "/home/pi/GassistPi/src/actions.py", line 99, in <module>
    pwm=GPIO.PWM(configuration['Gpios']['servo'], 50)
TypeError: an integer is required (got type list)
`

I don't have PulseAudio installed, but I've been using GassistPi without that being a problem.

shivasiddharth, 2018-08-23:
Small oversight, i have fixed the errata. Can you replace the files again and give it a shot. People get misguided with this pulse audio thing, infact there is no need for pulse audio. Its just a warning dont mind that.

georgefst, 2018-08-23:
Works, cheers.

Minor (weird) bug though: If I put either of the "assistant_indicators" values as "08" then it crashes with:
`Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 101, in <module>
    GPIO.setup(listening, GPIO.OUT)
ValueError: Channel must be an integer or list/tuple of integers
`

Works fine if I just put "8" but I initially went with "08" as that's how the other one digit numbers were formatted...

shivasiddharth, 2018-08-23:
Normally single digit GPIO numbers are not prefixed with 0.

PR #499 (2018-08-23): Disable ok google
Fixes #494
closed 2018-08-23 by shivasiddharth