shivasiddharth/GassistPi #496: Spotify

georgefst, 2018-08-22:
What is the likely future of Spotify integration in GassistPi? Would you be considering using spotipy to control spotify through the Web API (e.g. "play Losing My Religion" would use spotipy to play the song on the currently active Spotify Connect device)? This would presumably get around DRM issues, and in my personal use case this would work particularly well as I'm running [raspotify](https://github.com/dtcooper/raspotify) (a Spotify Connect client) so audio would actually play on the Pi itself.

If you're not interested in this, I may consider working on it myself, but I may not have the time right now...

PS great work so far - everything I wanted is working very smoothly, except playing music

shivasiddharth, 2018-08-23:
The spotify connect requires a premium/paid spotify subscription. I dont know as to how many users here would use that. Though having a separate music client is a great idea, there are other challenges associated. The google assistant should be able to interact with the music client (play/pause/volume controls). I am yet to find out a client that would do that.

Howardno, 2018-09-05:
What about Modipy for a spotify server? https://www.mopidy.com/

Looks like controlling volume and play/pause etc should work with Google assistant

georgefst, 2018-09-05:
Surprised I haven't seen that before, looks promising. I'll have a proper look this eve.

shivasiddharth, 2018-09-05:
We need the pi to act as a client more than a server.

tmxkn1, 2019-01-12:
+1 to this.

[**@Howardno**](https://github.com/Howardno) commented on [Sep 5, 2018, 10:27 AM GMT+1](https://github.com/shivasiddharth/GassistPi/issues/496#issuecomment-418663764 "2018-09-05T09:27:15Z - Replied by Github Reply Comments"):
> What about Modipy for a spotify server? [https://www.mopidy.com/](https://www.mopidy.com/)
> 
> Looks like controlling volume and play/pause etc should work with Google assistant

wouldn't this require another device to run Modipy as a spotify server?

jontyrudman, 2021-01-06:
I've just been looking into this a bit more, and it looks to be possible with spotipy (for using the API, premium feature as you said) and raspotify/librespot (for example if you're also running your pi as a Spotify Connect client).

If I have time this week I might look into integrating spotipy and raspotify with the media player to see if we can make calls to the Spotify API to play Spotify songs on the Pi using raspotify.

AryanRai, 2021-01-17:
one more thing can be done 
if you get the chromium drm version which works flawlessly on my pi 3 and somehow open chrome media edition (drm) version with https://open.spotify.com/search/song and here play it