shivasiddharth/GassistPi #497: Spotify doesn't work

pboucher44, 2018-08-23:
Hello everyone,
I hope you can help my with this bug:

When I say "play music on Spotify", I got a bug and I can't hear any music.

That is what my console says: 
```
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play music on Spotify"}
stopping vlc
Getting music links
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.23.8; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:01] Decoding of say.mp3 finished.
ERROR:__main__:[Errno 2] No such file or directory: '/tmp/say.mp3'
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 589, in <module>
    Myassistant().main()
  File "/home/pi/GassistPi/src/main.py", line 582, in main
    spotify_playlist_select(str(usrcmd).lower())
  File "/home/pi/GassistPi/src/actions.py", line 1309, in spotify_playlist_select
    say("Getting music links")
  File "/home/pi/GassistPi/src/actions.py", line 191, in say
    os.remove(ttsfilename)
FileNotFoundError: [Errno 2] No such file or directory: '/tmp/say.mp3'
^CException ignored in: <module 'threading' from '/usr/lib/python3.5/threading.py'>
Traceback (most recent call last):
  File "/usr/lib/python3.5/threading.py", line 1288, in _shutdown
    t.join()
  File "/usr/lib/python3.5/threading.py", line 1054, in join
    self._wait_for_tstate_lock()
  File "/usr/lib/python3.5/threading.py", line 1070, in _wait_for_tstate_lock
    elif lock.acquire(block, timeout):
```

Thank you for your help.

shivasiddharth, 2018-08-23:
This does not seem to follow the default spotify command syntax. Please go through the README for more details on the spotify action's capabilities.

pboucher44, 2018-08-23:
Thank you so much I now understand how it works :D
It was because I was trying to say the order in french.

closed 2018-08-23 by pboucher44