shivasiddharth/GassistPi #502: Unable to run assistant

LorenzoJenneker, 2018-08-25:
I run:
/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id 'replace this with the model id'
(with my model id, and quotes removed)

and get:
File "/home/pi/GassistPi/src/main.py", line 223 if event.type == EventType.ON_CONVERSATION_TURN_TIMEOUT or event.type == EventType.ON_NO_RESPONSE): ^ SyntaxError: invalid syntax

With ref to : #492
Still not working for me :'( please assist. I have the exact same error. At some point the start-up tone kept repeating in about 1mim intervals (don't know if it's related). Also "Ok google" does not work. Simply no response. No errors occurred at any point except as above. PLEASE PLEASE PLEASE assist :'( :'(

shivasiddharth, 2018-08-25:
Not sure as to when you cloned the repo. The above mentioned syntax error has long been fixed. Delete your existing GassistPi folder and clone the git again, no need to install the assistant again.

closed 2018-08-25 by shivasiddharth

LorenzoJenneker, 2018-08-26:
Thanks will try it soon just to make sure.
But i found another method of using simple usb mic i.e. "correcting" audio using the official voice hat image. But thank you for the response.