shivasiddharth/GassistPi #505: Custom wakeword not working

seasider1960, 2018-08-26:
### Describe the bug: The pre-installed custom wakewords (snowboy and alexa) do not work.  I noted that the config.yaml file has both "OK_Google" and "status" switches enabled by default contrary to the Youtube vid.
```

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No

## If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes - but only OK_Google works

### 4. Is the assistant service restarting automatically?
Yes

## If answer to question 4 is Yes, then paste the contents of the terminal below.

Not sure what you want here -- there's nothing outputted to the terminal when the service starts automatically.  See below for what's outputted on a manual start.  

```

## If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
Doesn't crash the assistant -- the OK_Google hotword works fine but not the  custom wakewords  See below.  A whole bunch of errors are thrown but as you can see OK_Google works fine

```

### Paste the contents of the terminal below

pi@PayPhonePi:~ $ /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id 'payphonepi-22ba6-payphone-pi-6c868o'
[01445190] pulse audio output error: PulseAudio server connection failure: Connection refused
[014515a8] pulse audio output error: PulseAudio server connection failure: Connection refused
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.front.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM front
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround40.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround40
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround41
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround50
[... 101 lines cut ...]
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "what time is it"}
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "what time is it"}
ON_RENDER_RESPONSE:
  {"text": "It's 6:09.", "type": 0}
ON_RENDER_RESPONSE:
  {"text": "It's 6:09.", "type": 0}
ON_RESPONDING_STARTED:
  {"is_error_response": false}
ON_RESPONDING_STARTED:
  {"is_error_response": false}
ON_RESPONDING_FINISHED
ON_RESPONDING_FINISHED
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}

```

### Attach the log file. Log file can be found in /tmp directory. Remeber to copy your log file before shutting down or restarting the Pi.

Sorry, could find any such file in /tmp

shivasiddharth, 2018-08-28:
I will change the "status" to something more relevant to avoid confusions. The OK Google has a better sensitivity that those provided custom wakewords. This might seem ridiculous but, here is my suggestion:
1. Get close to the mic and try different pronunciations of the wakeword. or
2. Create your own custom wakeword in the snowboy website and replace the existing models.

shivasiddharth, 2018-10-07:
Closing this for inactivity

closed 2018-10-07 by shivasiddharth