shivasiddharth/GassistPi #506: custom answer like google aiy kit

BarryAlvord, 2018-08-28:
Hi shiva,first of all i would like to say well done,i have done such a great work!!!!
second i just to ask you that con i creat a custom answer li google aiy voice kit.I have try using iftt to creat one but it just can creat only one so... Please answer me as soon as possible.

Thanks

shivasiddharth, 2018-08-28:
Do you have a link to the AIY Kit's custom conversation ? I will see what they have done and if possible release it in the next update.

BarryAlvord, 2018-08-29:
https://www.raspberrypi.org/forums/viewtopic.php?f=114&t=189054

BarryAlvord, 2018-08-29:
i hope this usefull::)

shivasiddharth, 2018-08-29:
Not a big  deal. Will add it in the next update.

BarryAlvord, 2018-08-30:
But just show us somewhere to edit to creat our own thanks

shivasiddharth, 2018-09-03:
https://github.com/shivasiddharth/GassistPi#custom-conversations Project has been updated with the ability to have custom conversations.

closed 2018-09-03 by shivasiddharth