shivasiddharth/GassistPi #508: spotify request quits gassistPi

pernordeman, 2018-08-28:
Hey! 

love the work!

Have bumped into i smal problem.. when requesting a playlist from spotify the code quits.. 

the following is printed when i request the playlist "work"
  {"text": "play work playlist from Spotify"}
stopping vlc
Getting music links
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.23.8; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:01] Decoding of say.mp3 finished.
work playlist

shivasiddharth, 2018-08-28:
You have got the syntax wrong. There should be no "playlist" word in the command.

pernordeman, 2018-08-28:
Okay thanks! The music is now playing but the code still quits when doing the request and the music start playing...

shivasiddharth, 2018-08-28:
Can you attach the log file ? It should be in the /tmp folder. And then post the output of manually run assistant.

pernordeman, 2018-08-28:
seems to be the spotify token..
The following is printed in tmp:

2018-08-28 17:20:02,574 ERROR __main__ name 'spotify_token' is not defined
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 516, in <module>
    main()
  File "/home/pi/GassistPi/src/main.py", line 510, in main
    spotify_playlist_select(str(usrcmd).lower())
  File "/home/pi/GassistPi/src/actions.py", line 1306, in spotify_playlist_select
    playlists,num=scan_spotify_playlists()
  File "/home/pi/GassistPi/src/actions.py", line 1254, in scan_spotify_playlists
    if spotify_token:
NameError: name 'spotify_token' is not defined

though i can still play specific songs... like if i say " play (song) from spotify" or "play (song) on chromecast"

shivasiddharth, 2018-08-28:
You need to add the spotify credentials in the config.yaml file. Casting to chromecast is a cloud feature that does not require custom actions, but assistant on pi does not support music streaming, so you need to use the custom action that i have prepd.

pernordeman, 2018-08-28:
Okay, the credentials in config.yaml seems to check with the credentials in action.py file.. 
Do i need the API for youtube and kickstarter aswell?

shivasiddharth, 2018-08-28:
Kickstarter uses the same api as youtube. And yes. I have clearly mentioned everything in the README doc.

pernordeman, 2018-08-28:
Is it any certain API key? have 4 different keys because im running the GassitPi with MagicMirror.. 
have API key 3 and 4, Browser key, and Server Key... should i create a new key?

shivasiddharth, 2018-08-28:
You generate an API key in the cloud console and then add the APIs that I can be used for. Say you have created an API key, now add search engine to the key ( for Kickstarter) and youtube data v3 (for YouTube). So it's basically the same key for both. 
I have added the same info in bold in README under both Kickstarter and Youtube actions options.

pernordeman, 2018-08-29:
does not seem to get it working... tried a new API key from console but it still quits:/
Can it be something else?

shivasiddharth, 2018-08-29:
Seems like a spotify credentials issue to me.

pernordeman, 2018-09-03:
do not seem to fix the problem.. but it something i could live with anyways... 
On another topic though..
Can the metricsystem be changed to Celius etc instead of fahrenheit?

shivasiddharth, 2018-12-16:
Units need to be changed in google's home app settings.

closed 2018-12-16 by shivasiddharth