shivasiddharth/GassistPi #509: Audio is not being recorded or playing on Raspberry PI Zero

vaddanki69, 2018-08-29:
### Describe the bug:
```
1. I bought a new Google AIY Voice kit from Target for $49.99. I believe it is google voice hat 2.0
2. Installed the packaged SD card and their GRPC demo works well.
3. I took a new SD card and installed 2018-06-27-raspbian-stretch.zip with Etcher.
4. Then after rebooting, executed the following commands as per GassistPi:
sudo apt-get install git  
git clone https://github.com/shivasiddharth/GassistPi  
sudo apt-get update  
sudo apt-get install raspberrypi-kernel 
Restart Pi
sudo chmod +x /home/pi/GassistPi/audio-drivers/AIY-HAT/scripts/configure-driver.sh  
sudo /home/pi/GassistPi/audio-drivers/AIY-HAT/scripts/configure-driver.sh  
sudo reboot  
sudo chmod +x /home/pi/GassistPi/audio-drivers/AIY-HAT/scripts/install-alsa-config.sh  
sudo /home/pi/GassistPi/audio-drivers/AIY-HAT/scripts/install-alsa-config.sh  
speaker-test  
Restart Pi
speaker-test -t wav  

No audio

Since, my PI is connected to TV using HDMI for display, Forced HDMI output
But No use.

Please help

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes/No

Yes

### 2. Is the issue related to assistant installation ?
Yes/No

Yes

## If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes/No

N/A

### 4. Is the assistant service restarting automatically?
Yes/No

N/A

## If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

### Attach the log file. Log file can be found in /tmp directory. Remeber to copy your log file before shutting down or restarting the Pi.

shivasiddharth, 2018-08-29:
Did you reboot after running the configure-driver.sh script ? If you did not reboot, then you will not hear the audio. 

Open a terminal and run
sudo nano /etc/asound.conf
And
sudo nano ~/.asoundrc

Check whether both the contents match and paste them here

shivasiddharth, 2018-10-07:
Closing this for inactivity

closed 2018-10-07 by shivasiddharth