shivasiddharth/GassistPi #510: language change for tasmota control

matrix1000, 2018-08-31:
Hi shivasiddharth,

Many thanks for your fantastic project.
Everything works nearly perfect.  :-)
I use gassistPi in German. For example I can ask for the time in german and I get a german answer.
Now I try to control tasmota devices. This works perfect too. I can switch on and off may lamps. :-)
But only in English. The command in my example is:

"Hallo google" "please turn on the Arbeitszimmer" Sounds funny, but it works.
Because in the /home/pi/GassistPi/src/config.yaml I have in friendly-names: "Arbeitszimmer"

The response from google is, "ok turning on working room" -> funny, so google translate my Arbeitszimmer->working room.

Also if I modify you main.py (def tasmota_control) exception: say("Device not online") to german -> "Gerät nicht erreichbar") google translate it to "device not available". ;-) 

Is there a way to control in completely in German?

Many thanks for you help

shivasiddharth, 2018-08-31:
Whatever you are hearing like "Turning on Lights" and "Device not online" are speech generated using a text to speech converter. By default, i have set the text to speech converter to English. You can change the language to German here https://github.com/shivasiddharth/GassistPi/blob/a3e5f45ac0cf4341a394756953212066efb552b3/src/actions.py#L151

matrix1000, 2018-08-31:
Hi,

You made my day. Now all works perfect.

For all others, you just have to change the english speeking words in the code too, then it works.

closed 2018-08-31 by shivasiddharth