shivasiddharth/GassistPi #512: GAssistPi Starting Muted

wastelandgunner, 2018-09-04:
### Describe the bug:
```
I've tried everything and cannot get GAssistPi to unmute itself. The button does nothing, and it always starts muted.
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No

## If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
No

### 4. Is the assistant service restarting automatically?
No

## If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```
/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device                                                                                                                               _model_id oceanassistant983-oceanassistant-fxibaj
INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
INFO:gmusicapi.Mobileclient1:version: 11.0.1
INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi                                                                                                                               .log
INFO:gmusicapi.Mobileclient1:initialized
INFO:gmusicapi.Mobileclient1:logged out
INFO:gmusicapi.Mobileclient1:failed to authenticate
[02971830] pulse audio output error: PulseAudio server connection failure: Conne                                                                                                                               ction refused
[0297da08] pulse audio output error: PulseAudio server connection failure: Conne                                                                                                                               ction refused
AIY-HAT
Initializing GPIO 25 for assistant activity indication
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.snd_r                                                                                                                               pi_googlev.pcm.front.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned err                                                                                                                               or: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directo                                                                                                                               ry
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM front
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.snd_r                                                                                                                               pi_googlev.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned err                                                                                                                               or: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directo                                                                                                                               ry
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.snd_r                                                                                                                               pi_googlev.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned err                                                                                                                               or: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directo                                                                                                                               ry
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.snd_r                                                                                                                               pi_googlev.pcm.surround40.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned err                                                                                                                               or: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directo                                                                                                                               ry
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround40
[... 30 lines cut ...]
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm_dmix.c:990:(snd_pcm_dmix_open) The dmix plugin supports only playba                                                                                                                               ck stream
ALSA lib pcm_dsnoop.c:556:(snd_pcm_dsnoop_open) The dsnoop plugin supports only                                                                                                                                capture stream
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unloc                                                                                                                               k
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unloc                                                                                                                               k
device_model_id: oceanassistant983-oceanassistant-fxibaj
device_id: 01B7323478AD501625C24A890BD09193

ON_MUTED_CHANGED:
  {"is_muted": true}
ON_MUTED_CHANGED:
  {"is_muted": true}
ON_START_FINISHED
ON_START_FINISHED

```

I wish I could attach a logfile but I don't seem to have one.
### Attach the log file. Log file can be found in /tmp directory. Remeber to copy your log file before shutting down or restarting the Pi.

shivasiddharth, 2018-09-04:
Open the /home/pi/ directory and choose to view hidden files and then delete the .mute file. Now if you restart your assistant it should start with mic turned on. The button does work, otherwise the assistant would not have muted in first place. Try playing around with the time in this line https://github.com/shivasiddharth/GassistPi/blob/02e348ed4e9b8528fce0a06399dd56890d839e4c/src/main.py#L197

shivasiddharth, 2018-10-07:
Closing this for inactivity

closed 2018-10-07 by shivasiddharth