shivasiddharth/GassistPi #513: Location name Bug

matrix1000, 2018-09-05:
Hi have a problem with locationnames.
If more than one are the same, the last char changes.
Version: v1.6.0-beta3
example devices.csv: 
```
1,"test01",192.168.0.254,admin,,bulb_1,aaaa
2,"test02",192.168.0.254,admin,,bulb_1,aaaa
3,"test03",192.168.0.254,admin,,bulb_1,aaaa
4,"test04",192.168.0.254,admin,,bulb_1,aaaa

```
Here a screenshot of the List
[image]

shivasiddharth, 2018-09-05:
What is this ? I dont think this is even a part of this project. Sorry I am unable to understand as to what you are trying to tell.

matrix1000, 2018-09-05:
Oh, I am so sorry, wrong git repo.

closed 2018-09-05 by matrix1000