shivasiddharth/GassistPi #518: authorization code issue?

mra1984, 2018-09-07:
https://developers.google.com/assistant/sdk/guides/library/python/embed/install-sample 
I am getting stuck on step 2

It should give me an authorisation code but for some reason is asking me for it as soon as paste in the path to json

I paste this in and modify:
google-oauthlib-tool --scope https://www.googleapis.com/auth/assistant-sdk-prototype \
      --scope https://www.googleapis.com/auth/gcm \
      --save --headless --client-secrets /path/to/client_secret_client-id.json

But it asks for authorization code right away?  Where can I find this?

Thanks

mra1984, 2018-09-10:
I mean I don't get a URL to visit it just asks for a code straight away?

shivasiddharth, 2018-09-10:
This forum is for issues pertaining to GassistPi alone. The instructions in the official website are kind of fragmented. That is why i have compiled everything together for this project users.

shivasiddharth, 2018-10-07:
Closing this for inactivity

closed 2018-10-07 by shivasiddharth