shivasiddharth/GassistPi #519: stream music problem

BarryAlvord, 2018-09-07:
here is my terminal output:
During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/__init__.py", line 41, in autodetect
    from . import file_cache
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/file_cache.py", line 41, in <module>
    'file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth')
ImportError: file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth
ERROR: Signature extraction failed: Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/youtube_dl/extractor/youtube.py", line 1210, in _decrypt_signature
    video_id, player_url, s
  File "/home/pi/env/lib/python3.5/site-packages/youtube_dl/extractor/youtube.py", line 1121, in _extract_signature_function
    res = self._parse_sig_js(code)
  File "/home/pi/env/lib/python3.5/site-packages/youtube_dl/extractor/youtube.py", line 1182, in _parse_sig_js
    jscode, 'Initial JS player signature function name', group='sig')
  File "/home/pi/env/lib/python3.5/site-packages/youtube_dl/extractor/common.py", line 972, in _search_regex
    raise RegexNotFoundError('Unable to extract %s' % _name)
youtube_dl.utils.RegexNotFoundError: Unable to extract Initial JS player signature function name; please report this issue on https://yt-dl.org/bug . Make sure you are using the latest version; see  https://yt-dl.org/update  on how to update. Be sure to call youtube-dl with the --verbose flag and include its complete output.
 (caused by RegexNotFoundError('Unable to extract \x1b[0;34mInitial JS player signature function name\x1b[0m; please report this issue on https://yt-dl.org/bug . Make sure you are using the latest version; see  https://yt-dl.org/update  on how to update. Be sure to call youtube-dl with the --verbose flag and include its complete output.',)); please report this issue on https://yt-dl.org/bug . Make sure you are using the latest version; see  https://yt-dl.org/update  on how to update. Be sure to call youtube-dl with the --verbose flag and include its complete output.
It is no longer available in youtube

Please help me

shivasiddharth, 2018-09-08:
What command did you issue ?

shivasiddharth, 2018-09-09:
I guess you changed it here https://github.com/shivasiddharth/GassistPi/blob/f3810228d3d2c0942deafc215eb3f950b299939e/src/main.py#L498 alone.
There are a couple of more places that you need to change then.
https://github.com/shivasiddharth/GassistPi/blob/f3810228d3d2c0942deafc215eb3f950b299939e/src/actions.py#L974
https://github.com/shivasiddharth/GassistPi/blob/f3810228d3d2c0942deafc215eb3f950b299939e/src/actions.py#L948

BarryAlvord, 2018-09-09:
i have done those things before 
i think the problem not ame from that

BarryAlvord, 2018-09-09:
here is the main problem:
ERROR: Signature extraction failed: Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/youtube_dl/extractor/youtube.py", line 1210, in _decrypt_signature
    video_id, player_url, s
  File "/home/pi/env/lib/python3.5/site-packages/youtube_dl/extractor/youtube.py", line 1121, in _extract_signature_function
    res = self._parse_sig_js(code)
  File "/home/pi/env/lib/python3.5/site-packages/youtube_dl/extractor/youtube.py", line 1182, in _parse_sig_js
    jscode, 'Initial JS player signature function name', group='sig')
  File "/home/pi/env/lib/python3.5/site-packages/youtube_dl/extractor/common.py", line 972, in _search_regex
    raise RegexNotFoundError('Unable to extract %s' % _name)
youtube_dl.utils.RegexNotFoundError: Unable to extract Initial JS player signature function name; please report this issue on https://yt-dl.org/bug . Make sure you are using the latest version; see  https://yt-dl.org/update  on how to update. Be sure to call youtube-dl with the --verbose flag and include its complete output.
 (caused by RegexNotFoundError('Unable to extract \x1b[0;34mInitial JS player signature function name\x1b[0m; please report this issue on https://yt-dl.org/bug . Make sure you are using the latest version; see  https://yt-dl.org/update  on how to update. Be sure to call youtube-dl with the --verbose flag and include its complete output.',)); please report this issue on https://yt-dl.org/bug . Make sure you are using the latest version; see  https://yt-dl.org/update  on how to update. Be sure to call youtube-dl with the --verbose flag and include its complete output.
It is no longer available in youtube

Maybe it the youtube-dl false

shivasiddharth, 2018-09-09:
This is why i ask people to use the issue template. Even in your previous issue #517  i said the same thing. I need to see the full command that you used, full terminal output including the command deciphered by the assistant .

BarryAlvord, 2018-09-09:
i just ask G.A to play baby im yours by saying "play baby im yours"

BarryAlvord, 2018-09-09:
everythings was normal when i make a few chang about the stream to play but now its getting error

shivasiddharth, 2018-09-09:
Whatever change that you have made has not been done properly. That is why you are getting 
> It is no longer available in youtube

BarryAlvord, 2018-09-09:
im reinstalling it