shivasiddharth/GassistPi #520: Audio Output Error only in performance, not in installation and test

tarepan, 2018-09-08:
Hi developer, thank you for great library.  

### Describe the bug:
Audio output is good when I install GassistPi (**speaker-test** and **speaker-test -t wav** work properly),
and final installation step work properly (I said "Hey Google, how is the weather?" to mic, and speaker said "today's weather is ...", work properly, great !!!!).  
But when I try **MANUALLY START THE ASSISTANT**, audio-related error occurred and no response.  
Even if I try **HEADLESS AUTOSTART on BOOT SERVICE SETUP**, same error seems to occur.

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No

### 3. Does the assistant service start normally?
No

### 4. Is the assistant service restarting automatically?
No

### Paste the command below, that crashed the assistant
```
/home/pi/env/bin/python -u /home/pi/GassistPi/src/pushbutton.py --project-id test-xxxxxxxx  --device-model-id test-yyyyyyyyyy
```

### Paste the contents of the terminal below
```
[01d552c8] pulse audio output error: PulseAudio server connection failure: Connection refused
[01d7cae0] pulse audio output error: PulseAudio server connection failure: Connection refused
USB-MIC-HDMI
Initializing GPIOs 5 and 6 for assistant activity indication
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/pushbutton.py", line 853, in <module>
    main()
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/GassistPi/src/pushbutton.py", line 707, in main
    flush_size=audio_flush_size
  File "/home/pi/env/lib/python3.5/site-packages/googlesamples/assistant/grpc/audio_helpers.py", line 190, in __init__
    blocksize=int(block_size/2),  # blocksize is in number of frames.
  File "/home/pi/env/lib/python3.5/site-packages/sounddevice.py", line 1264, in __init__
    **_remove_self(locals()))
  File "/home/pi/env/lib/python3.5/site-packages/sounddevice.py", line 684, in __init__
    samplerate)
  File "/home/pi/env/lib/python3.5/site-packages/sounddevice.py", line 2490, in _get_stream_parameters
    info = query_devices(device)
  File "/home/pi/env/lib/python3.5/site-packages/sounddevice.py", line 488, in query_devices
    raise PortAudioError('Error querying device {0}'.format(device))
sounddevice.PortAudioError: Error querying device -1
```

tarepan, 2018-09-08:
Sorry developer, It seems that this issue is duplicated to other issues (I make a mistake to find closed issues).

tarepan, 2018-09-08:
I checked "PulseAudio" related open/closed issues.  
This phenomenon is related to #482
In some issues you suggested that these PulseAudio message are only warning, not error.  
Unfortunately, in my case, these error stop program and fail to use GassitPi.  
So I reopen this issue.

shivasiddharth, 2018-09-08:
This is an issue with your  alsa config files. What is your audio setup ? check if your .asoundrc and asound.conf files are same. 
In the terminal type:
`sudo nano ~/.asoundrc`
then
`sudo nano /etc/asound.conf
`
check if both the files have same contents.

tarepan, 2018-09-09:
### sound setting
I do not configure any sound setting.  
I only install GassitPi into clean-installed raspberry pi Zero W.  

### .asoundrc & asound.conf
I checked the contents of ~/.asoundrc & /etc/asound.conf.  
They are exactly same.  
The contents is  
```
pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:1,0"
        channels 1
    }
}

pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "hw:0,1"
        }
        capture.pcm {
                type plug
                slave.pcm "dsnooper"
        }
}
```

### sound check
I checked RPiZW's audio itself, but it work properly (I can listen to YouTube.)  

### error change
I change the situation a little.  
I disable the headless start by **systemctl enable**, then try **MANUALLY START THE ASSISTANT** as before.  
resulting situation is not changed (in test ok, in performance ng), but error code change a little.  
Last three errors seems to be changed.  
final error is changed from  
```
sounddevice.PortAudioError: Error querying device -1
```
to
```
sounddevice.PortAudioError: Error opening RawStream: Device unavailable [PaErrorCode -9985]
```

Raw Error code is  
```
[02d66140] pulse audio output error: PulseAudio server connection failure: Connection refused
[02d4ee08] pulse audio output error: PulseAudio server connection failure: Connection refused
USB-MIC-HDMI
Initializing GPIOs 5 and 6 for assistant activity indication
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/pushbutton.py", line 853, in <module>
    main()
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/GassistPi/src/pushbutton.py", line 707, in main
    flush_size=audio_flush_size
  File "/home/pi/env/lib/python3.5/site-packages/googlesamples/assistant/grpc/audio_helpers.py", line 190, in __init__
    blocksize=int(block_size/2),  # blocksize is in number of frames.
  File "/home/pi/env/lib/python3.5/site-packages/sounddevice.py", line 1264, in __init__
    **_remove_self(locals()))

  File "/home/pi/env/lib/python3.5/site-packages/sounddevice.py", line 780, in __init__
    'Error opening {0}'.format(self.__class__.__name__))
  File "/home/pi/env/lib/python3.5/site-packages/sounddevice.py", line 2572, in _check
    raise PortAudioError(errormsg, err)
sounddevice.PortAudioError: Error opening RawStream: Device unavailable [PaErrorCode -9985]
```

**Error opening RawStream** is seen in #30, #181, #188, #356, but I cannot get suggestion...  

Thank you for helping me in your busy time!

shivasiddharth, 2018-09-09:
You have not answered all of my question. What is your audio setup? What mic and audio out device are you using?
open a terminal and run the following and paste the output here.
```
aplay -l
arecord -l
```

tarepan, 2018-09-10:
Sorry for bothering you.  
It seems that I do not understand the meaning of "audio setup"... but below information is ok?  

### mic and audio output device
* mic: USB-connected mic (ELEGIANTva6COuxel)  
* audio out device: HDMI-connected-monitor's built-in speaker

### aplay -l
```
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 7/7
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

### arecord -l
```
**** List of CAPTURE Hardware Devices ****
card 1: Device [USB PnP Sound Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

Thank you for your kind advice.

shivasiddharth, 2018-09-10:
Looks like the mic is not compatible. Can you try changing the mic ? You can even use a webcam as mic.

Arensfn, 2018-10-04:
Hello dear @shivasiddharth , I'm having the same issue but everything is the same in both files and my mic works also, could you please help me? 

_card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 7/7
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: H340 [Logitech USB Headset H340], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

card 1: H340 [Logitech USB Headset H340], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0_

Arensfn, 2018-10-04:
**HERE IS THE LOG**

INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
INFO:gmusicapi.Mobileclient1:version: 11.0.3
INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
INFO:gmusicapi.Mobileclient1:initialized
INFO:gmusicapi.Mobileclient1:logged out
INFO:gmusicapi.Mobileclient1:failed to authenticate
[02c999a0] pulse audio output error: PulseAudio server connection failure: Connection refused
[02cdc208] pulse audio output error: PulseAudio server connection failure: Connection refused
USB-MIC-JACK
Initializing GPIOs 5 and 6 for assistant activity indication
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.front.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM front
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround40.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround40
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround41
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround50
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround51
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround71.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround71
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM iec958
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
[... cut ...]
  File "/home/pi/GassistPi/src/main.py", line 157, in __init__
    self.detector = snowboydecoder.HotwordDetector(models, sensitivity=self.sensitivity)
  File "/home/pi/GassistPi/src/snowboydecoder.py", line 115, in __init__
    stream_callback=audio_callback)
  File "/home/pi/env/lib/python3.5/site-packages/pyaudio.py", line 750, in open
    stream = Stream(self, *args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/pyaudio.py", line 441, in __init__
    self._stream = pa.open(**arguments)
OSError: [Errno -9999] Unanticipated host error

shivasiddharth, 2018-10-05:
@Arensfn , You have got the config wrong. If you have a device with both audio and mic, you should use the USB DAC option.

Arensfn, 2018-10-05:
@shivasiddharth , Thanks for replying, I tried USB DAC but I'm getting the same error. 
should I re-Install the GassistPi?

shivasiddharth, 2018-10-07:
What version of the OS are you running ?

Arensfn, 2018-10-08:
@shivasiddharth please help I tried changing the rates and other stuff but didn't work could the issue caused by the Microphone?

shivasiddharth, 2018-10-08:
Changing stuff randomly will worsen the situation it will get tougher for me to troubleshoot. This is clearly an issue with the microphone.

Arensfn, 2018-10-08:
I put back everything on default. So what do you think about my issue. Any suggestion?

Arensfn, 2018-10-10:
@shivasiddharth did you find any solution for me?

shivasiddharth, 2018-10-10:
I have already said that this could be an issue with the Mic. Try changing the Mic. If you change the audio setup, you need to change the config files as well.

karolosz, 2018-11-01:
hello ,  i got trouble with
**[pulseaudio] alsa-util.c: Unable to load mixer: Invalid argument**

when i run  commend
systemctl status gassistpi-ok-google.service
 i get
* gassistpi-ok-google.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi-ok-google.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2018-11-01 00:13:05 CET; 11h ago
 Main PID: 419 (python)
   CGroup: /system.slice/gassistpi-ok-google.service
           |-419 /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id saved-model-id
           `-929 /usr/bin/pulseaudio --start --log-target=syslog

Nov 01 00:13:25 raspberrypi python[419]:     register this device follow instructions at:
Nov 01 00:13:25 raspberrypi python[419]:     https://developers.google.com/assistant/sdk/guides/library/python/embed/registe
Nov 01 00:13:26 raspberrypi python[419]: ON_MUTED_CHANGED:
Nov 01 00:13:26 raspberrypi python[419]:   {"is_muted": false}
Nov 01 00:13:26 raspberrypi python[419]: ON_MUTED_CHANGED:
Nov 01 00:13:26 raspberrypi python[419]:   {"is_muted": false}
Nov 01 00:13:26 raspberrypi python[419]: ON_START_FINISHED
Nov 01 00:13:26 raspberrypi python[419]: ON_START_FINISHED
Nov 01 00:13:56 raspberrypi pulseaudio[929]: [pulseaudio] alsa-util.c: Unable to load mixer: Invalid argument
Nov 01 00:13:56 raspberrypi pulseaudio[929]: [pulseaudio] alsa-util.c: Unable to load mixer: Invalid argument

i runnig raspberry pi 3 zith mic from camera PS3 and speaker connected to minijack

and my setting of asound.conf and .asoundrc are the same:

pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:1,0"
        channels 1
    }
}

pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "hw:0,0"
        }
        capture.pcm {
                type plug
                slave.pcm "dsnooper"
        }
}

thanx for help
KaRol

shivasiddharth, 2018-12-16:
Closing this for inactivity.

closed 2018-12-16 by shivasiddharth

jolimk, 2019-07-07:
Hey, Im getting the error 
`sounddevice.PortAudioError: Error opening RawStream: Invalid sample rate [PaErrorCode -9997]
`
When I try to 
`googlesamples-assistant-pushtotalk --project-id XXX --device-model-id XXX --display`