shivasiddharth/GassistPi #521: youtube streaming

ulugeyik, 2018-09-10:
### Describe the bug:
```
I use youtube streaming regularly but it has started to fail two days ago. It produces the usual "getting youtube link" or the weird noise , waits a bit, rings chimes and says "hi". I have tried resetting the raspberry pi (using with googlevoice AIY), I have looked at the logs etc.  The only odd thing I have noticed is that if I manually download a particular video, then it works.  Everything else seem to work fine.

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No

## If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
No. I do not think so.

### Attach the log file. Log file can be found in /tmp directory. Remeber to copy your log file before shutting down or restarting the Pi.
I have copied two log files. One "sil.txt" is what happens when a previously unplayed track is requested by saying "stream abba" which fails to play, the other *sil2.txt) is a previously downloaded one (manually by youtube-dl) which works by saying "stream c is for cookie". Both files were generated using " tail -f GassistPi.log > sil.txt" before saying "hey google, stream ..."

[sil.txt](https://github.com/shivasiddharth/GassistPi/files/2367405/sil.txt)
[sil2.txt](https://github.com/shivasiddharth/GassistPi/files/2367406/sil2.txt)

shivasiddharth, 2018-09-10:
Open a terminal type:
```
source env/bin/activate
pip install --upgrade youtube_dl
```
After that restart your assistant.

ulugeyik, 2018-09-11:
Thank you, works. I did not appreciate the need for sourcing the environment and the need for pip.  tried it with apt-get and did not find and upgrade. Thanks very much.

ulugeyik, 2018-09-22:
The problem is back and upgrading is not working. 
Could it be that my gassistpi scripts are outdated? I am a bit vary about saying "git pull" since I am not sure what happens to the customized configurations. Is that what I should do?

shivasiddharth, 2018-09-22:
Manually run the assistant and post the output.

shivasiddharth, 2018-10-07:
Closing this for inactivity

closed 2018-10-07 by shivasiddharth

ulugeyik, 2018-10-18:
I apologize. I have been traveling and did not have access to the unit. Do you mean that I should run "assistant_library_demo.py" ? 

Incidentally, how do I upgrade the gassistpi scripts? If I git clone , would it overwrite my configuration options?

aandroide, 2019-05-17:
hello, it happened to me after 2 months that I didn't update youtube.
It would not be a bad idea to integrate an automatic system that updates youtube into the project when it becomes obsolete.
@shivasiddharth 
what do you think about it