shivasiddharth/GassistPi #524: Control LED strip when Assistant is talkink

Fringo9, 2018-09-13:
Hi, is it possible to control a LED strip in order to make the LED flashing when the assistant is talking?

seasider1960, 2018-09-22:
I've just done this using a WS2801 strip.  I'm sure there are more elegant approaches but what I did was simply connect the "listening" and "speaking"  GPIO outputs (the defaults in the config file are GPIOs 5 and 6) to two other GPIOs configured as inputs.  To reduce the chance for a mishap I placed a 2K resistor in series between each input and output.  Then all you need are a couple of simple functions to listen for the input event and include those in your script running the led strip:

def ga_Listening():
    if (GPIO.input(gaListening) == True) and (GPIO.input(gaResponding) == False):
        for i in range(32):
            pixels.set_pixel_rgb(i, 255, 0, 0)
        pixels.show()       
                 
def ga_Responding():
    if (GPIO.input(gaResponding) == True) and (GPIO.input(gaListening) == False):
        for i in range(32):
            pixels.set_pixel_rgb(i, 0, 0, 255)
        pixels.show()

Here the functions simply turn all the pixels to red and blue to indicate the listening and speaking but of course you could insert any effect you wanted.

shivasiddharth, 2019-01-06:
Closing this for inactivity.

closed 2019-01-06 by shivasiddharth