shivasiddharth/GassistPi #526: custom wakeword not working

mra1984, 2018-09-14:
### Describe the bug:
```
Have followed custom wakeword activation instructions but Alexa does not work, nor does my custom *.pmdl snowboy file.  Any ideas please?  I tried searching old issues but everything is enabled and files are loaded in the right location. 
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No

## If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
Yes

## If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

### Attach the log file. Log file can be found in /tmp directory. Remeber to copy your log file before shutting down or restarting the Pi.

Pealy, 2018-09-15:
I thought I had the same problem until I tried speaking right next to my USB mic and Alexa was recognised.
I had set ok_google to Disabled and commented out the snowboy.umdl line

shivasiddharth, 2018-09-15:
Duplicate of #505

closed 2018-09-15 by shivasiddharth