shivasiddharth/GassistPi #527: power controls not working on pi

mra1984, 2018-09-14:
### Describe the bug:
```
I have tried saying -- trigger, shutdown and power off.....

However, google says 'sorry power controls is not supported?' 

what have I done wrong?  Thanks 
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes
### 2. Is the issue related to assistant installation ?
Unsure

## If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
Yes/No

## If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

### Attach the log file. Log file can be found in /tmp directory. Remeber to copy your log file before shutting down or restarting the Pi.

shivasiddharth, 2018-09-14:
Already discussed in #419  #251  #199  and many more.. It is not picking up the keyword "trigger".

mra1984, 2018-09-18:
I have spent a long time looking at these posts and others, please excuse my lack of experience.  However, the only actionable piece I could find was to change 'shut down' to 'shutdown' in actions.py

When I say "ok google shutdown", it just says power controls not supported

mra1984, 2018-09-18:
it also doesn't autostart for me either; here is my code - again I have read all the issues on this`[Unit]
Description=Google Assistant-Voice
Wants=network-online.target
After=network-online.target

[Service]
Environment=VIRTUAL_ENV=/home/pi/env
Environment=PATH=/home/pi/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id  'voicekit-212614' --device_model_id 'voicekit-212614-friday2-eqgema'

WorkingDirectory=/home/pi/
StandardOutput=inherit
StandardError=inherit
Restart=always
User=pi

[Install]
WantedBy=multi-user.target
`

shivasiddharth, 2018-09-22:
@mra1984 Its not picking up the keyword "trigger" is guess change it to something else in the kedwords.yaml file. If it manually starts, then autostart should workonly reason it would not happen. open a terminal and type:
```
sudo systemctl stop gassistpi-ok-google.service
sudo systemctl disable gassistpi-ok-google.service
```
Then go through the auto starting procedure once again.

shivasiddharth, 2018-11-12:
Closing this for inactivity

closed 2018-11-12 by shivasiddharth