shivasiddharth/GassistPi #532: FATAL start-loop with "fcm_client_impl.cc"

vvallee, 2018-09-16:
### Describe the bug:
```
I have a Pi3 B with the AIY kit.
After install the gassistpi worked well for about 10 minutes, then suddenly it went into an error state and continues to restart until stopped. A reboot does not resolve the issue.
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes, no issues found

### 2. Is the issue related to assistant installation ?
No

## If answer to question 2 is Yes, then paste the contents of the terminal below.

```
```

## If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes - Sometimes

### 4. Is the assistant service restarting automatically?
Yes

## If answer to question 4 is Yes, then paste the contents of the terminal below.

```
Sep 16 22:10:26 controller python[1988]: ALSA lib pcm_dmix.c:990:(snd_pcm_dmix_open) The dmix plugin supports only playback stream
Sep 16 22:10:26 controller python[1988]: ALSA lib pcm_dsnoop.c:556:(snd_pcm_dsnoop_open) The dsnoop plugin supports only capture stream
Sep 16 22:10:26 controller python[1988]: Cannot connect to server socket err = No such file or directory
Sep 16 22:10:26 controller python[1988]: Cannot connect to server request channel
Sep 16 22:10:26 controller python[1988]: jack server is not running or cannot be started
Sep 16 22:10:26 controller python[1988]: JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Sep 16 22:10:26 controller python[1988]: JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Sep 16 22:10:26 controller python[1988]: device_model_id: saved-model-id
Sep 16 22:10:26 controller python[1988]: device_id: CF65F9C55647909C153323AFE032ACEC
Sep 16 22:10:26 controller python[1988]:     This device is not registered. This means you will not be able to use
Sep 16 22:10:26 controller python[1988]:     Device Actions or see your device in Assistant Settings. In order to
Sep 16 22:10:26 controller python[1988]:     register this device follow instructions at:
Sep 16 22:10:26 controller python[1988]:     https://developers.google.com/assistant/sdk/guides/library/python/embed/register-device
Sep 16 22:10:27 controller python[1988]: ON_MUTED_CHANGED:
Sep 16 22:10:27 controller python[1988]:   {"is_muted": false}
Sep 16 22:10:27 controller python[1988]: ON_MUTED_CHANGED:
Sep 16 22:10:27 controller python[1988]:   {"is_muted": false}
Sep 16 22:10:27 controller python[1988]: ON_START_FINISHED
Sep 16 22:10:27 controller python[1988]: ON_START_FINISHED
Sep 16 22:10:28 controller python[1988]: [1988:2027:FATAL:fcm_client_impl.cc(382)] Check failed: fcm_store_->AddIidToken(app_id, registration_request->sender_id, iid_token).
Sep 16 22:10:28 controller systemd[1]: gassistpi-ok-google.service: Main process exited, code=killed, status=6/ABRT
Sep 16 22:10:28 controller systemd[1]: gassistpi-ok-google.service: Unit entered failed state.
Sep 16 22:10:28 controller systemd[1]: gassistpi-ok-google.service: Failed with result 'signal'.
Sep 16 22:10:28 controller systemd[1]: gassistpi-ok-google.service: Service hold-off time over, scheduling restart.
Sep 16 22:10:28 controller systemd[1]: Stopped Google Assistant-Voice.
```

## If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

### Attach the log file. Log file can be found in /tmp directory. Remeber to copy your log file before shutting down or restarting the Pi.

shivasiddharth, 2018-09-16:
Can you manually start the assistant and post the output

vvallee, 2018-09-17:
This seems to have fixed the problem thanks!

closed 2018-09-17 by vvallee