shivasiddharth/GassistPi #534: error that i dont even know what to call

BarryAlvord, 2018-09-21:
here is my terminal output:
ON_MUTED_CHANGED:
  {"is_muted": false}
ON_MUTED_CHANGED:
  {"is_muted": false}
ON_START_FINISHED
ON_START_FINISHED
INFO:snowboy:Keyword 1 detected at time: 2018-09-21 03:33:55
Assistant is listening....
ON_CONVERSATION_TURN_STARTED

ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play tear in my Heart by Twenty One Pilots"}
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play tear in my Heart by Twenty One Pilots"}
stopping vlc
ERROR:__main__:'NoneType' object has no attribute 'group'
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/actions.py", line 998, in YouTube_No_Autoplay
    say("Getting youtube link")
  File "/home/pi/GassistPi/src/actions.py", line 201, in say
    words= translator.translate(words, dest=language)
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/client.py", line 172, in translate
    data = self._translate(text, dest, src)
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/client.py", line 75, in _translate
    token = self.token_acquirer.do(text)
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/gtoken.py", line 180, in do
    self._update()
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/gtoken.py", line 59, in _update
    code = unicode(self.RE_TKK.search(r.text).group(1)).replace('var ', '')
AttributeError: 'NoneType' object has no attribute 'group'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 650, in <module>
    Myassistant().main()
  File "/home/pi/GassistPi/src/main.py", line 516, in main
    YouTube_No_Autoplay(str(usrcmd).lower())
  File "/home/pi/GassistPi/src/actions.py", line 1011, in YouTube_No_Autoplay
    say('It is no longer available in youtube')
  File "/home/pi/GassistPi/src/actions.py", line 201, in say
    words= translator.translate(words, dest=language)
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/client.py", line 172, in translate
    data = self._translate(text, dest, src)
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/client.py", line 75, in _translate
    token = self.token_acquirer.do(text)
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/gtoken.py", line 180, in do
    self._update()
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/gtoken.py", line 59, in _update
    code = unicode(self.RE_TKK.search(r.text).group(1)).replace('var ', '')
AttributeError: 'NoneType' object has no attribute 'group'
INFO:snowboy:Keyword 1 detected at time: 2018-09-21 03:34:45
^CException ignored in: <module 'threading' from '/usr/lib/python3.5/threading.py'>
Traceback (most recent call last):
  File "/usr/lib/python3.5/threading.py", line 1288, in _shutdown
    t.join()
  File "/usr/lib/python3.5/threading.py", line 1054, in join
    self._wait_for_tstate_lock()
  File "/usr/lib/python3.5/threading.py", line 1070, in _wait_for_tstate_lock
    elif lock.acquire(block, timeout):

i think we should blame google

IngmarG, 2018-09-21:
It started to appear yesterday and seems to have something to do with 503 returns from Google that require you to "demonstrate" that you're not a robot.

shivasiddharth, 2018-09-21:
The google translate package has to be update. once this is merged, the google translate package has to be updated manually.

IngmarG, 2018-09-21:
Didn't work for me. Installed patched version in site-packages, still same error.
But then I am not a Python programmer. Did I do it correctly? (I moved all of the GoogleTrans py files to an "old" directory, then unpacked the Googletrans folder contents from the patched version at https://github.com/isdaves/py-meowbot into your original folder.

Is that the way to do it (temporarily)? Or did I forget something?

I had to do this because I can't wait to make my project work. It did work yesterday morning, but after working all day to put it in a box, this error appeared :-(.

shivasiddharth, 2018-09-21:
Open the environment folder, packages then google translate, there you should find a file called gtoken.py, replace that with this [raw file] (https://github.com/BoseCorp/py-googletrans/blob/360f6027a613cabb57da93a861e750f48db572fd/googletrans/gtoken.py)

BarryAlvord, 2018-09-21:
it just does not work:
ON_MUTED_CHANGED:
  {"is_muted": false}
ON_MUTED_CHANGED:
  {"is_muted": false}
ON_START_FINISHED
ON_START_FINISHED
INFO:snowboy:Keyword 1 detected at time: 2018-09-21 11:17:03
Assistant is listening....
ON_CONVERSATION_TURN_STARTED

ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "please turn off the table light"}
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "please turn off the table light"}
Turning Off table light
ERROR:__main__:'NoneType' object has no attribute 'group'
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 650, in <module>
    Myassistant().main()
  File "/home/pi/GassistPi/src/main.py", line 509, in main
    Action(str(usrcmd).lower())
  File "/home/pi/GassistPi/src/actions.py", line 1411, in Action
    say("Turning Off " + name)
  File "/home/pi/GassistPi/src/actions.py", line 207, in say
    tts.save(ttsfilename)
  File "/home/pi/env/lib/python3.5/site-packages/gtts/tts.py", line 247, in save
    self.write_to_fp(f)
  File "/home/pi/env/lib/python3.5/site-packages/gtts/tts.py", line 187, in write_to_fp
    part_tk = self.token.calculate_token(part)
  File "/home/pi/env/lib/python3.5/site-packages/gtts_token/gtts_token.py", line 28, in calculate_token
    seed = self._get_token_key()
  File "/home/pi/env/lib/python3.5/site-packages/gtts_token/gtts_token.py", line 62, in _get_token_key
    a = re.search("a\\\\x3d(-?\d+);", tkk_expr).group(1)
AttributeError: 'NoneType' object has no attribute 'group'

BarryAlvord, 2018-09-21:
sir IngmarG u know hwo to connect camera to GassistPi?

shivasiddharth, 2018-09-21:
Its a multipoint error, thanks to Google's mods. After fixing the translator token like how i mentioned previously you need to fix the gtts token as well. Check  [this](https://github.com/pndurette/gTTS/issues/137#issuecomment-423480337)

IngmarG, 2018-09-21:
Yes, just found that also. Will try it in an hour or so.

IngmarG, 2018-09-21:
@BarryAlvord,

Yes, install the picamera module in the environment. I'll send you the code in a PM

BarryAlvord, 2018-09-21:
ITs working mr.shiva

BarryAlvord, 2018-09-21:
could u so me your work sir IngmarG

BarryAlvord, 2018-09-21:
Thanks for showing me @IngmarG

BarryAlvord, 2018-09-21:
sr but PM stand for what sir @IngmarG 

p/s sr for my eng

IngmarG, 2018-09-21:
No problem. PM: personal message, as it does not belong in this thread.

BarryAlvord, 2018-09-21:
please send it to my email sir @IngmarG

IngmarG, 2018-09-21:
> my email:[bibi221991@gmail.com](mailto:bibi221991@gmail.com) sỉr @IngmarG

Done. I suggest you remove your email from the thread.

aandroide, 2018-09-23:
Hello to all,
I updated the project today starting from scratch, and I was presented with the same error.
As a first error, the LED of the AIY box does not light up, the log file is no longer created and finally
I followed the advice of sid and I noticed that 2 days ago they released gTTS-token 1.1.2 which I installed manually but the error is:
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "stream blablabla"}
stopping vlc
ERROR: __ main __: 'NoneType' object has no attribute 'group'
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/actions.py", line 1004, in YouTube_No_Autoplay
    say ("Getting youtube link")
  File "/home/pi/GassistPi/src/actions.py", line 207, in say
    words = translator.translate (words, dest = language)
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/client.py", line 132, in trad
    data = self._translate (text, dest, src)
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/client.py", line 57, in _translate
    token = self.token_acquirer.do (text)
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/gtoken.py", line 180, in do
    self._update ()
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/gtoken.py", line 59, in _update
    code = unicode (self.RE_TKK.search (r.text) .group (1)). replace ('var', '')
AttributeError: 'NoneType' object has no attribute 'group'

During the exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 661, in <module>
    Myassistant (). Main ()
  File "/home/pi/GassistPi/src/main.py", line 518, in main
    YouTube_No_Autoplay (str (usrcmd) flower setting ())
  File "/home/pi/GassistPi/src/actions.py", line 1017, in YouTube_No_Autoplay
    say ('It is no longer available in youtube')
  File "/home/pi/GassistPi/src/actions.py", line 207, in say
    words = translator.translate (words, dest = language)
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/client.py", line 132, in trad
    data = self._translate (text, dest, src)
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/client.py", line 57, in _translate
    token = self.token_acquirer.do (text)
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/gtoken.py", line 180, in do
    self._update ()
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/gtoken.py", line 59, in _update
    code = unicode (self.RE_TKK.search (r.text) .group (1)). replace ('var', '')
AttributeError: 'NoneType' object has no attribute 'group'

aandroide, 2018-09-23:
I just solved the problem of the LED that did not light up on AIY-HAT. the installation made from scratch, had not created the audiosetup file.
I reinstalled the audio drivers and the file appeared in home / pi and now the LED works. only the error of the stream that reports the errors mentioned above remains

shivasiddharth, 2018-09-23:
@aandroide i would suggest you to manually change and fix the files how i have mentioned above.

aandroide, 2018-09-24:
@shivasiddharth 
thanks sid, I was just wrong to edit the file instead of python 3.5 I edited the file in /home/pi/.local/lib/python2.7/site-packages/gtts_token/gtts_token.py.
Is the gassistpi.log file in the temp folder available to you? because I do not see it anymore after the update

NeoMod, 2018-10-07:
On Pi3 - Raspbian Stretch Lite:

I had to manually install gTTS using 

`pip install gTTS`

The gTTS file then is located in

> /home/pi/env/lib/python3.5/site-packages/gtts_token

but there is no need to edit it.
_

Hoeby, 2018-10-11:
Does somebody have an idea what to do?

Changed to 3 lines to 6, didn't worked (hope i changed i correct)
I have done a pip install gTTS, also didn't worked.
Found some more information and more options what there needed to be changed instead of adding the 6 lines. But also that didn't work.

I don't know where to look now, where it is going wrong.

NeoMod, 2018-10-11:
@Hoeby From my experience, I would suggest starting with a fresh install (both SO and GassistPI) and then installing latest gTTS version via pip.

For me this solved the issue without further editing.

Don't forget to fix the gtoken.py file as explained previously by  shivasiddharth.

Hoeby, 2018-10-12:
@NeoMod,
3 days ago i started with a fresh install. (every day a few steps where done)
Went step by step whit the manual from Shivasiddharth. 
Yesterday i got it working and next step was to enabled domoticz function, and that was the moment i got the problem.

Only thing to do a fresh install could be, that i bought a Pi SD card with NOOBS already installed. Maybe start over with the raspbian image. I didn't done this because update said i already had all updates

NeoMod, 2018-10-12:
@Hoeby mmmm....yes, starting with a fresh raspbian installation would have been my next move too. I would suggest trying to start with RASPBIAN LITE if you don't plan on using desktop functionalities on your raspberry. A part from the first impact (you just deal with terminal, it's a bit disarming for the first minutes but then I assure you it becomes easy and natural) this setup grants the bare minimum installed, so you'll benefit also in terms of performances. (more resources available for GassistPI).

Anyway, whichever version you need I would suggest doing a fresh install and start again from scratch. 
Then If the problem presents only upon enabling Domoticz functionalities you'll have a definitive "point of origin' for the problem.
I'll try to work with Domoticz in the evening, maybe I could find something helpful. (Nothing granted tough, I'm merely a noob with some experience.😋 )

Hoeby, 2018-10-12:
thanks Shivasiddharth,
I replaced the gtoken.py and give it a go.
The error i had is gone, but now the say.mp3 file can't be found.
Just tried it in a few minutes i had, now need to look further what's went wrong whit the mp3 file. Already thanks for all the help

`FileNotFoundError: [Errno 2] No such file or directory: '/tmp/say.mp3'
`

Hoeby, 2018-10-12:
Before shutdown my PI, i tried it again.
And now it works perfect. Maybe the first time it neede to write the mp3 file.
Now no errors. Tonight i will test it with speakers connected, this was a quick test without them

Hoeby, 2018-10-12:
Just tested what is going on.
I run GassistPi as service. When starting up the PI and run it as service. Then it works perfect.
When i try to start it manual. Then it is started twice. The service and the manual is started. When i ask something like the time, then i get 2 answers. The manual started GassistPi stops working after the first question, it gives the answer that it can't find /tmp/say.mp3 
The service started GassistPi keeps working perfect. Gives me a big smile on the face.
Don't know if it is good by now, but i am happy for now. Hope the errors stay away when trying more options in the future.

A very very big thank you to Shivasiddharth, you made wishes come true.

oceank82, 2018-11-06:
Hello!

I have installed this from scratch and I cannot get it to work. What do I have to do? I get this error:

nov 06 20:09:35 raspberrypi python[4802]:     transword= translator.translate(words, dest=lang)
nov 06 20:09:35 raspberrypi python[4802]:   File "/home/pi/env/lib/python3.5/site-packages/googletrans/client.py", line 
nov 06 20:09:35 raspberrypi python[4802]:     data = self._translate(text, dest, src)
nov 06 20:09:35 raspberrypi python[4802]:   File "/home/pi/env/lib/python3.5/site-packages/googletrans/client.py", line 
nov 06 20:09:35 raspberrypi python[4802]:     token = self.token_acquirer.do(text)
nov 06 20:09:35 raspberrypi python[4802]:   File "/home/pi/env/lib/python3.5/site-packages/googletrans/gtoken.py", line 
nov 06 20:09:35 raspberrypi python[4802]:     self._update()
nov 06 20:09:35 raspberrypi python[4802]:   File "/home/pi/env/lib/python3.5/site-packages/googletrans/gtoken.py", line 
nov 06 20:09:35 raspberrypi python[4802]:     code = unicode(self.RE_TKK.search(r.text).group(1)).replace('var ', '')
nov 06 20:09:35 raspberrypi python[4802]: AttributeError: 'NoneType' object has no attribute 'group'

adelorenzo, 2018-11-17:
Dear Shivasiddharth,

First and foremost I would like to thank you for your project. I did everything that has been suggested, checked my Deezer ID, downloaded the gtoken.py, edited the gtts_token.py file, fresh install and unfortunately I still cannot make my GA work with the customizations, unfortunately. It always freezes somehow. Please advise. Thank you.

AltOfCtrl, 2018-11-17:
> The last line provide a possible hint for the solution:
> 
> `gtts_token File "/home/pi/env/lib/python3.5/site-packages/gtts_token/gtts_token.py", line 68 lineStr = line ^ IndentationError: unexpected indent`
> 
> There could be a typo in the gtts_token.py file.
> 
> (Did you edited that file? Because now you need just to install the latest gTTS package and then edit the gtoken.py inside /home/pi/env/lib/python3.5/site-packages/googletrans )

Ah, that seems to be it. Lucky I copied the file to my desktop before I edited it. Everything seems to be in working order now. Many Thanks.

edit: ok, everything seems to work - until I ask one of the default custom response questions 'who are you?' at which point it stops replying to my questions (I assume a crash occurs). starting the service manually gives results but the output shows two results to my question,

`ON_CONVERSATION_TURN_STARTED

ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "what time is it"}
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "what time is it"}
ON_RENDER_RESPONSE:
  {"text": "7:34.", "type": 0}
ON_RENDER_RESPONSE:
  {"text": "7:34.", "type": 0}
ON_RESPONDING_STARTED:
  {"is_error_response": false}
ON_RESPONDING_STARTED:
  {"is_error_response": false}
ON_RESPONDING_FINISHED
ON_RESPONDING_FINISHED
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}
`

 and when I close the terminal I stop receiving answers (potentially showing that the auto start doesn't work but the manual does) but obviously me continually starting the service manually isn't sustainable

NeoMod, 2018-11-17:
This happens - most likely - because the gTTS engine is not installed.

Stop the assistant and issue `pip install gTTS` then reload the assistant.

When the assistant is started manually:

- output can be read in the console, and if no other problem occurs spoken output occurs from the device
- the assistant will work up until you leave the terminal open (because the way it has started)
- it is good practice, if you have configured the auto-start service for GassistPI, to stop the assistant before starting again in terminal mode.
- when you close the terminal the auto-start feature will not kick-in because it is not supposed to: the auto-start service works upon system boot.
- The commands at your disposal are `sudo service gassistpi-ok-google start` and `sudo service gassistpi-ok-google stop`. (for example: you issue the stop command before running the terminal instance, then after closing the terminal istance you issue the start command to relaunch the service, which will then be operating in background)

AltOfCtrl, 2018-11-17:
> pip install gTTS

Ok, I stopped the assistant using 'sudo service gassistpi-ok-google stop' installed gTTS then used  '/home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --device_model_id [my info redacted]' to start the service in a command window to view any issues.

when I asked 'who are you' I received this

ON_CONVERSATION_TURN_STARTED

ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "who are you"}
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "who are you"}
ERROR:__main__:name 'Keyerror' is not defined
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 470, in main
    say(selectedans[0])
  File "/home/pi/GassistPi/src/actions.py", line 219, in say
    newword=trans(words,language)
  File "/home/pi/GassistPi/src/actions.py", line 210, in trans
    transword= translator.translate(words, dest=lang)
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/client.py", line 172, in translate
    data = self._translate(text, dest, src)
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/client.py", line 75, in _translate
    token = self.token_acquirer.do(text)
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/gtoken.py", line 180, in do
    self._update()
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/gtoken.py", line 59, in _update
    code = unicode(self.RE_TKK.search(r.text).group(1)).replace('var ', '')
AttributeError: 'NoneType' object has no attribute 'group'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 673, in <module>
    Myassistant().main()
  File "/home/pi/GassistPi/src/main.py", line 472, in main
    except Keyerror:
NameError: name 'Keyerror' is not defined

(It also starts off with the same double written response thing)

NeoMod, 2018-11-17:
It seems there is an error in your `gtoken.py` file.

Can you remove your version and replace with the (RAW) version obtained from here?

https://github.com/BoseCorp/py-googletrans/blob/360f6027a613cabb57da93a861e750f48db572fd/googletrans/gtoken.py

I just tried conversation: the two set of question/answers overlaps with default Google answers. Most likely you will obtain the response written in `config.yaml` only when asking in english.

AltOfCtrl, 2018-11-17:
I replaced my file with the code from the raw version. I also added 2 custom responses to test if it will work, I added my google api key so I can use the youtube add-on and I changed the alexa wake word to smart_mirror. I started the service using `/home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --device_model_id [my info redacted]`

Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 40, in <module>
    from actions import say
  File "/home/pi/GassistPi/src/actions.py", line 15, in <module>
    from mediaplayer import api
  File "/home/pi/GassistPi/src/mediaplayer.py", line 9, in <module>
    from youtube_search_engine import youtube_search
  File "/home/pi/GassistPi/src/youtube_search_engine.py", line 13, in <module>
    configuration = yaml.load(conf)
  File "/home/pi/env/lib/python3.5/site-packages/yaml/__init__.py", line 72, in load
    return loader.get_single_data()
  File "/home/pi/env/lib/python3.5/site-packages/yaml/constructor.py", line 35, in get_single_data
    node = self.get_single_node()
  File "/home/pi/env/lib/python3.5/site-packages/yaml/composer.py", line 36, in get_single_node
    document = self.compose_document()
  File "/home/pi/env/lib/python3.5/site-packages/yaml/composer.py", line 55, in compose_document
    node = self.compose_node(None, None)
  File "/home/pi/env/lib/python3.5/site-packages/yaml/composer.py", line 84, in compose_node
    node = self.compose_mapping_node(anchor)
  File "/home/pi/env/lib/python3.5/site-packages/yaml/composer.py", line 133, in compose_mapping_node
    item_value = self.compose_node(node, item_key)
  File "/home/pi/env/lib/python3.5/site-packages/yaml/composer.py", line 84, in compose_node
    node = self.compose_mapping_node(anchor)
  File "/home/pi/env/lib/python3.5/site-packages/yaml/composer.py", line 133, in compose_mapping_node
    item_value = self.compose_node(node, item_key)
  File "/home/pi/env/lib/python3.5/site-packages/yaml/composer.py", line 84, in compose_node
    node = self.compose_mapping_node(anchor)
  File "/home/pi/env/lib/python3.5/site-packages/yaml/composer.py", line 133, in compose_mapping_node
    item_value = self.compose_node(node, item_key)
  File "/home/pi/env/lib/python3.5/site-packages/yaml/composer.py", line 82, in compose_node
    node = self.compose_sequence_node(anchor)
  File "/home/pi/env/lib/python3.5/site-packages/yaml/composer.py", line 110, in compose_sequence_node
    while not self.check_event(SequenceEndEvent):
  File "/home/pi/env/lib/python3.5/site-packages/yaml/parser.py", line 98, in check_event
    self.current_event = self.state()
  File "/home/pi/env/lib/python3.5/site-packages/yaml/parser.py", line 393, in parse_block_sequence_entry
    "expected <block end>, but found %r" % token.id, token.start_mark)
yaml.parser.ParserError: while parsing a block collection
  in "/home/pi/GassistPi/src/config.yaml", line 12, column 7
expected <block end>, but found '<scalar>'
  in "/home/pi/GassistPi/src/config.yaml", line 12, column 31

not quite sure

NeoMod, 2018-11-17:
Ok, now this appears to be an error with your custom response in config.yaml.

May I suggest to do one thing at a time? 

1) Start with the default config.yaml, and see if everything works.
2) Add your custom response and test it to see if it's working properly and if the syntax is right 
3) Add your YouTube API KEY and test it. Also, remember to actually activate the YouTube Data API in the Cloud Console
4) Modify the wakeword and test it.

Anyway, this is no longer an issue related to the google translate token.

AltOfCtrl, 2018-11-18:
> Ok, now this appears to be an error with your custom response in config.yaml.
>
> May I suggest to do one thing at a time?
>
> 1. Start with the default config.yaml, and see if everything works.
> 2. Add [...]

Ok, it turned out the main issue stemmed from me using a apostrophe in my custom response. Everything does seem to be working now. (I am having some trouble trying to change the wake word, but that's a different issue to the google translate token as such I will take it elsewhere) thanks for the help

DewGew, 2018-11-29:
I have updated gtoken.py and run  `pip instal gtts` but i still got errors:
```
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "who are you"}
ERROR:__main__:name 'Keyerror' is not defined
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 489, in main
    say(selectedans[0])
  File "/home/pi/GassistPi/src/actions.py", line 227, in say
    newword=trans(words,language)
  File "/home/pi/GassistPi/src/actions.py", line 218, in trans
    transword= translator.translate(words, dest=lang)
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/client.py", line 172, in translate
    data = self._translate(text, dest, src)
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/client.py", line 75, in _translate
    token = self.token_acquirer.do(text)
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/gtoken.py", line 187, in do
    self._update()
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/gtoken.py", line 66, in _update
    code = unicode(self.RE_TKK.search(r.text).group(1)).replace('var ', '')
AttributeError: 'NoneType' object has no attribute 'group'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 692, in <module>
    Myassistant().main()
  File "/home/pi/GassistPi/src/main.py", line 491, in main
    except Keyerror:
NameError: name 'Keyerror' is not defined
```
Any clues?

shivasiddharth, 2018-11-29:
Can you double check if gtoken has been replaced correctly. GTTS should be updated within the environment. Just running pip install will not work. Use:
```
source env/bin/activate
pip install gTTS
```

Hoeby, 2018-11-29:
Have the same problem over here. The day before yesterday it worked fine. Now i get exactly the same error as DewGew has. 
Checked the gtoken.py. Just to be sure, i copied the raw gtoken.py from this Https://raw.githubusercontent.com/BoseCorp/py-googletrans/360f6027a613cabb57da93a861e750f48db572fd/googletrans/gtoken.py and put it in the gtoken.py on my pi. But this makes no difference.

After that did this, but also this makes no difference
```
source env/bin/activate
pip install gTTS
```

Any ideas?

shivasiddharth, 2018-11-29:
I have updated the links to the new token in https://github.com/shivasiddharth/GassistPi/issues/534#issuecomment-423495674 sometime back. Make sure to replace the correct token file. For an easy way out,
```
source env/bin/activate
pip install goslate

```
Replace your existing actions.py with the new file from the project master (https://raw.githubusercontent.com/shivasiddharth/GassistPi/master/src/actions.py)

I have updated the project to get rid of this error, that is why this issue was closed in the last commit.

shivasiddharth, 2018-11-30:
@Hoeby , the file you replaced is wrong. The correct file is : https://raw.githubusercontent.com/BoseCorp/py-googletrans/506ec8e5a161045b1effaffec565b1a704e392c8/googletrans/gtoken.py

DewGew, 2018-11-30:
Hmmm ..
 Now I got this ..
`ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play Radio 1"}
http://stream.100p.nl/100pctnl.mp3
ERROR:__main__:HTTP Error 503: Service Unavailable
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 692, in <module>
    Myassistant().main()
  File "/home/pi/GassistPi/src/main.py", line 554, in main
    radio(str(usrcmd).lower())
  File "/home/pi/GassistPi/src/actions.py", line 280, in radio
    say("Tuning into " + conv)
  File "/home/pi/GassistPi/src/actions.py", line 225, in say
    newword=trans(words,language)
  File "/home/pi/GassistPi/src/actions.py", line 219, in trans
    transword= translator.translate(words, lang)
  File "/home/pi/env/lib/python3.5/site-packages/goslate.py", line 409, in translate
    return _unwrapper_single_element(self._translate_single_text(text, target_language, source_language))
  File "/home/pi/env/lib/python3.5/site-packages/goslate.py", line 334, in _translate_single_text
    results = list(self._execute(make_task(i) for i in split_text(text)))
  File "/home/pi/env/lib/python3.5/site-packages/goslate.py", line 203, in _execute
    yield each()
  File "/home/pi/env/lib/python3.5/site-packages/goslate.py", line 332, in <lambda>
    return lambda: self._basic_translate(text, target_language, source_lauguage)[0]
  File "/home/pi/env/lib/python3.5/site-packages/goslate.py", line 251, in _basic_translate
    response_content = self._open_url(url)
  File "/home/pi/env/lib/python3.5/site-packages/goslate.py", line 191, in _open_url
    raise e
  File "/home/pi/env/lib/python3.5/site-packages/goslate.py", line 181, in _open_url
    response = self._opener.open(request, timeout=self._TIMEOUT)
  File "/usr/lib/python3.5/urllib/request.py", line 472, in open
    response = meth(req, response)
  File "/usr/lib/python3.5/urllib/request.py", line 582, in http_response
    'http', request, response, code, msg, hdrs)
  File "/usr/lib/python3.5/urllib/request.py", line 510, in error
    return self._call_chain(*args)
  File "/usr/lib/python3.5/urllib/request.py", line 444, in _call_chain
    result = func(*args)
  File "/usr/lib/python3.5/urllib/request.py", line 590, in http_error_default
    raise HTTPError(req.full_url, code, msg, hdrs, fp)
urllib.error.HTTPError: HTTP Error 503: Service Unavailable`

shivasiddharth, 2018-11-30:
@DewGew This must be with the goslate. I have reverted back to googletranslate. Please go through https://github.com/shivasiddharth/GassistPi/issues/534#issuecomment-442942748

GZubieta, 2018-12-02:
you install 
$ pip uninstall googletrans
$ git clone https://github.com/BoseCorp/py-googletrans.git
$ cd ./py-googletrans
$ python setup.py install

ykr412, 2018-12-09:
anything i ask in the config.yamil i get an error
ON_MUTED_CHANGED:
  {"is_muted": false}
ON_MUTED_CHANGED:
  {"is_muted": false}
ON_START_FINISHED
ON_START_FINISHED
ON_CONVERSATION_TURN_STARTED

ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "who are you"}
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "who are you"}
Who do you think I am
ERROR:__main__:name 'Keyerror' is not defined
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 489, in main
    say(selectedans[0])
  File "/home/pi/GassistPi/src/actions.py", line 230, in say
    tts = gTTS(text=newword, lang=language)
  File "/home/pi/env/lib/python3.5/site-packages/gtts/tts.py", line 119, in __init__
    raise ValueError("Language not supported: %s" % lang)
ValueError: Language not supported: en

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 692, in <module>
    Myassistant().main()
  File "/home/pi/GassistPi/src/main.py", line 491, in main
    except Keyerror:
NameError: name 'Keyerror' is not defined

shivasiddharth, 2018-12-09:
@ykr412 , Your issue is not the same as the one listed here, open up a new issue or edit your existing issue. From the post here, seems like the mic is working properly, so i would suggest you to edit the issue #680 that you have already opened.