shivasiddharth/GassistPi #538: Suggestion: move functionalities outside of main

IngmarG, 2018-09-22:
This program is getting more and more functionalities, but not everybody wants everything.

Wouldn't it be an idea to move all functionalities outside the main.py part so they can be imported on a per case basis? I e.g. had only a need for a voice controlled camera, so I deleted everything in main and actions that I couldn't use, except for the custom words in the "for event in ...." loop, and then added the "camera" word part.

shivasiddharth, 2018-09-22:
Its actually WIP. Not only is it easy for users, its easy for me as well, instead of duplicating the functions in both pushbutton and main scripts i can update it in only one place.

IngmarG, 2018-09-22:
True, but as it grows, lots of functionality (and code) goes in, which to some may be totally irrelevant.

Anyway, I use your code, and my camera works perfectly. (and on top the standard Google actions are accesible, ;-) )

shivasiddharth, 2018-12-14:
Can you check if this branch works as it should do ? 
https://github.com/shivasiddharth/GassistPi/tree/clean-up
I have moved everything out of main.py to actions.py.