shivasiddharth/GassistPi #539: Can't manually start the assistant

Boai12, 2018-09-22:
### Describe the bug:
```
After following the installation everything works fine. The assistant works and I can talk to it. But after stopping the assistant and trying to manually start up it wont work.
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No

## If answer to question 2 is Yes, then paste the contents of the terminal below.

```
pi@raspberrypi:~ $ /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id 'gassistant-71851'
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 31, in <module>
    import snowboydecoder
  File "/home/pi/GassistPi/src/snowboydecoder.py", line 5, in <module>
    import snowboydetect
  File "/home/pi/GassistPi/src/snowboydetect.py", line 18, in <module>
    _snowboydetect = swig_import_helper()
  File "/home/pi/GassistPi/src/snowboydetect.py", line 15, in swig_import_helper
    return importlib.import_module(mname)
  File "/usr/lib/python3.5/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xa8 in position 2: invalid start byte

```

## If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
No

### 4. Is the assistant service restarting automatically?
No

## If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

### Attach the log file. Log file can be found in /tmp directory. Remeber to copy your log file before shutting down or restarting the Pi.

shivasiddharth, 2018-10-07:
Which version of the OS are you running ?

shivasiddharth, 2018-11-11:
Closing this for inactivity.

closed 2018-11-11 by shivasiddharth