shivasiddharth/GassistPi #544: Pushbutton does not work

aandroide, 2018-09-24:
### Describe the bug:
```
I have configured the trigger pin in config.yaml but nothing happens when the button is pressed.
how much I try to close with ctrl + c it starts to listen in loop
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No

## If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
No

## If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```
[02d41c38] pulse audio output error: PulseAudio server connection failure: Connection refused
[02d1cd20] pulse audio output error: PulseAudio server connection failure: Connection refused
AIY-HAT
Initializing GPIO 25 for assistant activity indication

```

### Attach the log file. Log file can be found in /tmp directory. Remeber to copy your log file before shutting down or restarting the Pi.

moreover, the log file is not created

shivasiddharth, 2018-09-25:
You need to disable custom wakeword for pushbutton to work. Its either or. Either the pushbutton trigger or custom wakeword. You cannot have both.

aandroide, 2018-09-25:
Wow, if I had read it in the readme I would not have opened the problem.

shivasiddharth, 2018-09-25:
I thought i had it in the README. Thanks for notifying i have now added it under the custom wakewords section. https://github.com/shivasiddharth/GassistPi/blob/master/README.md#custom-wakeword-activation

shivasiddharth, 2018-09-27:
I guess we lost the log after the custom wakeword thing. Thats because we are not catching errors from threads. Will fix it in the next update.

fragolinux, 2018-12-30:
> I guess we lost the log after the custom wakeword thing. Thats because we are not catching errors from threads. Will fix it in the next update.

still no logs in gassistpi folder

shivasiddharth, 2019-01-16:
@fragolinux , @aandroide Has fixed this now.

closed 2019-01-16 by shivasiddharth