shivasiddharth/GassistPi #545: Manual start of assistant gives Pulseaudio error

matrix98053, 2018-09-26:
### Describe the bug:
```
If i mual start the assistant i get an error about Pulseaudio server connection failed: connection refused. I know Pulseaudio is not installed. If i install it, it works for a few minutes then i get no sound or mic input. If i uninstall the audio and mic comes back, but the Assistant cannot start beacause of the Pulseaudio error. Is there a file i can edit to change Pulseaudio to Alsa?
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No

## If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes

### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```
pi@raspberrypi:~ $ /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --d                            evice_model_id 'linen-setting-216308-googlepi-15xb6g'
INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
INFO:gmusicapi.Mobileclient1:version: 11.0.3
INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
INFO:gmusicapi.Mobileclient1:initialized
INFO:gmusicapi.Mobileclient1:logged out
INFO:gmusicapi.Mobileclient1:failed to authenticate
[0192ff08] pulse audio output error: PulseAudio server connection failure: Connection refused
[01919b60] pulse audio output error: PulseAudio server connection failure: Connection refused
Initializing GPIOs 5 and 6 for assistant activity indication
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
[FATAL:audio_input_stream.cc(47)] Input device could not be opened: default
Aborted
pi@raspberrypi:~ $
```

### Paste the contents of the terminal below
```

```

### Attach the log file. Log file can be found in /tmp directory. Remeber to copy your log file before shutting down or restarting the Pi.

shivasiddharth, 2018-09-26:
Pulseaudio error has already been discussed in #494 #451 #449 #416 #413 . 

> [FATAL:audio_input_stream.cc(47)] Input device could not be opened: default
Aborted

is because of improperly configured audio scripts. What is your audio setup ? What mic and audio out are you using?

matrix98053, 2018-09-26:
Thanks for the reply.
I am using HDMI audio on the RPi3 and I have a 4Mic aray Seeed hat.
I am still a novice...

shivasiddharth, 2018-09-26:
Looks like you have used the Respeaker's official guidelines to setup the HAT. The default alsa files for Respeaker HAT does not work for this project. So i have modified them to suit this project. Remove your existing Respeaker HAT installation and follow 3.6 under https://github.com/shivasiddharth/GassistPi#install-audio-config-files For best results, the 4 mic HAT is set to work with 3.5mm Jack and not HDMI.

matrix98053, 2018-09-26:
That sorted it out. 
Thank you very much.
I did not know there was a diffrence between the two sets of drivers for Respeaker.

closed 2018-09-26 by matrix98053

shivasiddharth, 2018-09-26:
Out of box, their scripts dont work with this project and also i had to modify their scripts to make the onboard LEDs work.